motioneye-project/motioneye #1467: Grabbing the pixel count deltas

stryngs, 2019-09-25:
Can you point me to the part of the code that dictates the Integer represented in the top right of the video output showing the (pixel changes?) please?

I'm trying to take that integer and do things with it in code on the system the motioneye server runs on.

Any help you can provide would be much much appreciated!

zagrim, 2019-09-25:
The numbers you are talking about is what Motion puts in the frames when https://motion-project.github.io/motion_config.html#text_changes is enabled in the camera config, so that integer is never available in MotionEye except in the image data.

stryngs, 2019-09-26:
I'm sure there is a way to grab it as an object.  Thank you for pointing me in the right direction.  If I figure it out I'll make sure to post here.

jasaw, 2019-09-26:
@stryngs You could modify [motion](https://github.com/Motion-Project/motion) software to write it to a file, and you inotify that file.

JamBalaya56562, 2026-11-10:
Closing, as this was answered: that number is drawn into the picture by Motion's `text_changes` option and never reaches motionEye, so getting it as a value means changing Motion itself, as jasaw suggested.

closed 2026-11-10 by JamBalaya56562