motioneye-project/motioneye #1469: No video feed from a USB cam

GuyZipory, 2019-09-26:
Just got MotionEye on a Pi 3 and connected a Logitech C930e usb camera.
as the MotionOS loads and I enter the local website to see the feed from the camera,
now I can see that the camera is detected but all I see is a gray box, no video feed.

What could be the problem?

Barmatuhin, 2019-09-26:
Check if Streaming enabled

GuyZipory, 2019-09-26:
It's enabled
it's detect as:
Camera Device: /dev/v4l/by-id/usb-046d_Logitech_Webcam_C930e_601C53AE-video-index0
Camera Type: V4L2 Camera

GuyZipory, 2019-09-26:
Okey, Update:
If I go to the addres of the stream I can see the video feed good.
But at the motioneye website where is there is all the settings and small video feed of the cameras, there at that web site the feed is all gray, and becuase of it I can't make masks. or anything

CRImier, 2019-09-26:
Hi! First of all, any chance you're using Google Chrome? It recently updated and broke the technique MotionEye uses for webpage image streaming. You can read #1401 to know more. Tl;dr: you can use another browser that didn't make the same change (i.e. Firefox, Edge, or some Webkit browsers that didn't yet implement this same change), update MotionEye to the latest release, or use the about://flags fix that's described in #1401 .

GuyZipory, 2019-09-26:
> Hi! First of all, any chance you're using Google Chrome? It recently updated and broke the technique MotionEye uses for webpage image streaming. You can read #1401 to know more. Tl;dr: you can use [...]

@CRImier 
OMG Yes! It's fixed the problem, I was useing Chrome and now I checked of Firefox and everything working great!
Thanks!

closed 2019-09-26 by GuyZipory