motioneye-project/motioneye #1470: LDAP for user login

deide6102, 2019-09-26:
It would be awesome to be able to have users login using LDAP. I am planning on using motion eye in a large environment and I was hoping to be able to have multiple user accounts and to be able to use LDAP for authentication. This way I could set permissions etc.

deide6102, 2019-09-26:
Or if anyone has a way of changing the .html files in the template to .php files so that I can use my own authentication.

zagrim, 2019-09-26:
Multi-user support was just asked in #1436, and refused. Sorry.

deide6102, 2019-09-26:
> Multi-user support was just asked in #1436, and refused. Sorry.

Do you know of a way to be able to add support for it?

zagrim, 2019-09-27:
Calin Crisan decides what to do with MotionEye as it is his project, and I don't know what would happen if someone created a pull request to add LDAP and/or multi-user support (or maybe it would be better if that someone volunteered to do that and would discuss the architectural decisions that need to be made along the way with @ccrisan ). Myself I'm not familiar enough with LDAP to be willing to try regardless if support for it was wanted or not, the little effort I can give to this project will go to things that are smaller scale.

If you have some idea how to piggy-back your own authentication, I think you'd need to start from the source code, do whatever needs to be done with the templates and then build your own version of MotionEye. But I honestly don't know how that would work out :)

ccrisan, 2019-09-27:
Ideally 3rd party authentication mechanisms would be completely separated from motionEye. Users would have to map to normal or administrator rights in the end. User account management is not and shouldn't be motionEye's concern.

One way to go about this is to proxy everything through a piece of software that handles any kind of sophisticated authentication and forwards the requests in the name of the administrator or normal user to motionEye. This piece of software could also handle access logs, failed auth logging & alike.

Long story short, LDAP authentication is not going to be part of motionEye.

closed 2019-09-27 by ccrisan

dmak, 2019-11-11:
I agree that adding all kind of authentication methods and protocols is cumbersome. And ti can be done on e.g. Apache level. But how to implement that? Suppose that Apache has authenticated the user and can e.g. set the `X-Forwarded-User` header when proxying the request to MotionEye. Is MotionEye ready for this sort of interaction? I.e. is there any flag or option to skip authentication step (= don't show login dialog) if certain conditions are met?

I would suggest the following setup:
```
<Location "/motioneye">                               
        RewriteEngine on                              
        RewriteRule /motioneye$         /motioneye/ [R,L]
        RewriteRule /motioneye/(.*)$    http://localhost:8765/$1 [P,L]
        ...
        Require valid-user                            
                                              
        RequestHeader unset X-Forwarded-User          
        RequestHeader set X-Forwarded-User admin
</Location>
```
To solve the issue #1323 (map user `batman` to `admin`) one can say:
```
        RequestHeader set X-Forwarded-User admin "expr=%{REMOTE_USER} == 'batman'"
```
Similarly one can map users to `guest` account however I am not sure how to map e.g. LDAP group (see [this post](http://apache-http-server.18135.x6.nabble.com/mod-authnz-ldap-and-env-vars-td4747699.html)), probably by putting `SetEnv` into `AuthzProviderAlias`?

P.S. See similar setup for [Jenkins](https://wiki.jenkins.io/display/JENKINS/Reverse+Proxy+Auth+Plugin#ReverseProxyAuthPlugin-ApacheConfigurationExample).