motioneye-project/motioneye #1477: Pycurl problems ?

GevaudanBeast, 2019-09-30:
Hello,
I have this message : 
`óè motioneye.service - motionEye Server                                                                     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)                   Active: failed (Result: exit-code) since Mon 2019-09-30 21:39:51 BST; 4s ago                             Process: 1804 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited Main PID: 1804 (code=exited, status=1/FAILURE)                                                                                                                                                                     Sep 30 21:39:51 Ha-Bridge meyectl[1804]:     import motionctl                                             Sep 30 21:39:51 Ha-Bridge meyectl[1804]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctSep 30 21:39:51 Ha-Bridge meyectl[1804]:     import mediafiles                                            Sep 30 21:39:51 Ha-Bridge meyectl[1804]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/mediafilSep 30 21:39:51 Ha-Bridge meyectl[1804]:     import uploadservices                                        Sep 30 21:39:51 Ha-Bridge meyectl[1804]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadseSep 30 21:39:51 Ha-Bridge meyectl[1804]:     import pycurl                                                Sep 30 21:39:51 Ha-Bridge meyectl[1804]: ImportError: /usr/lib/arm-linux-gnueabihf/libcurl.so.4: version `Sep 30 21:39:51 Ha-Bridge systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURSep 30 21:39:51 Ha-Bridge systemd[1]: motioneye.service: Failed with result 'exit-code'.`

No success with reinstall « pip uninstall / install picurl »
Can you help me please ?

GevaudanBeast, 2019-10-03:
Can someone help me?
I do not know if it's important, but it's a Raspberry Pi 2 on which a HA-Bridge server is installed and that uses Raspbian Buster

JamBalaya56562, 2026-11-08:
Closing, as the traceback ends with `import pycurl` failing with `ImportError: /usr/lib/arm-linux-gnueabihf/libcurl.so.4: version …`: the pycurl of the Python 2 install had been built for another libcurl than the one of Raspbian Buster, as in #2585. Current releases run on Python 3 and install pycurl with pip in a virtual environment, see the [README](https://github.com/motioneye-project/motioneye#installation).

closed 2026-11-08 by JamBalaya56562