motioneye-project/motioneye #1479: Home recognition for MotionEye(OS) with Fritz!Box using fritzconnection (python)

da-dindi, 2019-10-02:
Hi!

Not much to say here as everything is explained in code comments.

This uses fritzconnection that can be found [here](https://github.com/kbr/fritzconnection)

Here is the code:
```
#!/usr/bin/env python
#coding: utf-8

#This script is launched at boot via cron and runs on my server pi
#(Raspberry Pi 3b+) where it switches MotionEyeOS motion
#detection on another pi (Raspberry pi Zero W) depending on if
#given devices are found in the Fritz!Box (WIFI)network.

#Be aware of that this is a beginners work. If you find anything
#that could be improved feel free to tell me.
#I am from germany and therefore my english is not quite good.
#Sorry ;-)

import os
import time
from credentials import * #Import login data stored in separate file
#(could for sure be improved a lot)

motionstatus = "UNSET" #Workaround to only check status of motion
#detection on first cycle (to minimize network traffic of this script)

#create pid file
pid = str(os.getpid())
f = open('/home/pi/fbhomerecognition_pid', 'w') #Change path to fit your needs
f.write(pid)
f.close()

while True:
    try:
        #Read data from Fritz!Box with fritzconnection and check if given MAC addresses stored in credentials.py are online
        home = 0
        online = os.popen ("python3 /home/pi/fritzconnection/fritzconnection/fritzwlan.py -i 192.168.178.1 -u %s -p %s" % (fbuser, fbpass)).read() #Path depends on where you installed fritzconnection. IP needs to be set to Fritz!Box IP. fbuser and fpbass are stored in credentials.py
        for n in maclist:
            result = online.find(n)
            if result != -1:
                home = 1
            else:
                pass

        #Read status of motion detection from MotionEye(OS)
        if motionstatus == "UNSET":
            status = os.popen("curl http://192.168.178.2:7999/1/detection/status").read() #Change IP to where your MotionEye(OS) is running. Port is standard port. "1" is the number of the camera to check. For more information read MotionEye(OS) wiki
            if status.find("PAUSE")  != -1:
                motionstatus = "PAUSE"
            elif status.find("ACTIVE") != -1:
                motionstatus = "ACTIVE"
            else:
                pass
        else:
            pass

        #Switch motion detection to "ACTIVE" if not already active as none of the devices from maclist are online in Fritz!Box WIFI
        if home == 0:
            if motionstatus == "PAUSE":
                os.system ("curl http://192.168.178.2:7999/1/detection/start") #See line 38
                motionstatus = "ACTIVE"
            else:
                pass
        #Switch motion detection to "PAUSE" if not already paused as one or more devices from maclist are online in Fritz!Box WIFI
        elif home == 1:
            if motionstatus == "ACTIVE":
                os.system ("curl http://192.168.178.2:7999/1/detection/pause") #See line 38
                motionstatus = "PAUSE"
            else:
                pass

        #Wait for x seconds before this runs again
        time.sleep(60)
    except:
        exit()
```

Credentials file for login data:
```
#!/usr/bin/python
# coding: utf-8

 #List of MAC addresses that stop motion detection if they are detected online in the home WIFI
maclist = ("00:00:00:00:00:00", "00:00:00:00:00:00", "00:00:00:00:00:00")

#The user you add here needs some configuration in Fritz!Box. For further info see the fritzconnection documentation.
fbuser = "fbuser" #Fritz!Box username
fbpass = "fbpass" #Fritz!Box password
```

Files for download:
[fbhomerecognition.zip](https://github.com/ccrisan/motioneye/files/3680986/fbhomerecognition.zip)

Hope someone finds this helpfull.

Firesphere, 2021-04-26:
I took your python script, and did some little changes to it, to make it work a bit smoother, and, especially on RPi's, reduce the system load when someone is home.
You can find it here:
https://github.com/Firesphere/motioneye-fritz-home

Firesphere, 2021-04-27:
@da-dindi I'm no Python expert either :) But I hope my slight improvements help you learn to get better at it!

JamBalaya56562, 2026-11-13:
Thanks for sharing. @Firesphere's improved version is at https://github.com/Firesphere/motioneye-fritz-home. Since motionEye 0.45.0b1, such a script can also use motionEye's own `POST /action/<camera id>/detectionpause` and `detectionstart` (see #251), which work where Motion's control port only listens locally. Closing, as there is nothing to change in motionEye itself.

closed 2026-11-13 by JamBalaya56562