motioneye-project/motioneye #1480: Motion detection occasionally stops in running docker container

JBS5, 2019-10-03:
At the moment I am running 0.41 in docker on a NUC using Ubuntu.
Some days ago, motion detection has stopped working without any apparent reason. I am not able to get the motion detection to work. Restarting the container did not fix it.
The camera stream is available in motioneye, only the motion detection part is not working anymore.

When logging in into the docker container en tailing motion.log, this is what I see:

```
root@motioneye:/var/log# tail motion.log
[1:ml1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera((null)): Connection refused
[1:ml1] [ERR] [VID] vid_start: Netcam RTSP failed to open
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera((null)): Connection refused
[1:ml1] [ERR] [VID] vid_start: Netcam RTSP failed to open
[1:ml1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera((null)): Connection refused
[1:ml1] [ERR] [VID] vid_start: Netcam RTSP failed to open
```

Any clue what's happening?

JBS5, 2019-11-03:
Motion detecting has stopped again without any reason (not that I know off). Does anyone recognize this?

JBS5, 2019-12-05:
Installed motionEye on my NUC this way https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu instead of using the docker, and motion detection is still working..

JamBalaya56562, 2026-11-07:
Closing, as motion detection keeps working for you since installing motionEye directly instead of in Docker, as with #1444. The cause in the container stayed unknown; please open a new issue if it happens with a current image.

closed 2026-11-07 by JamBalaya56562