motioneye-project/motioneye #1482: Nginx tutorial not working without base_path

stefmitropoulos, 2019-10-05:
Hello!

I followed the tutorial to the 't' multiple times in order to run my motioned app behind Nginx.

In the instructions we would  create a location /cams/ which will be proxy_passed to a an http://ip:port/.

According to the Nginx documentation, if we specify a uri to the proxied server (in this case /), then ourserver/cams/ should reverse proxy to http://ip:port/. 

My issue is that this is not happening and I get the default index.html "Welcome to NGINX server" message instead.

`server {
    listen 443 ssl;
    server_name motioneye.*;
    include /config/nginx/ssl.conf;
    client_max_body_size 0;
    # enable for ldap auth, fill in ldap details in ldap.conf
    #include /config/nginx/ldap.conf;
    
location /cams/ {
        # enable the next two lines for http auth
        include /config/nginx/proxy.conf;
        proxy_pass http://MYINTERNALIP:8765/;
    }
}`

Am I misunderstanding something?

thnurg, 2019-10-07:
With the proxy in place your new URL should be https://MYINTERNALIP/cams/

Let me know if it works. I'm trying to get reverse proxy via apache working but I think I have to set the base URL somewhere.

JamBalaya56562, 2026-11-10:
Closing, as this is in the nginx setup: nginx picks the `server` block by the requested host name, so with `server_name motioneye.*`, opening the IP address lands on the default server and its welcome page. Opening `https://motioneye.<domain>/cams/` reaches motionEye, which needs no base path for this.

closed 2026-11-10 by JamBalaya56562