motioneye-project/motioneye #1483: Fedora 30 "Exit-code" / "ImportError: pycurl: libcurl link-time ssl backend (openssl) is different from compile-time ssl backend (none/other)"

Mytherium, 2019-10-06:
Followed the instructions in the Fedora guide to the letter, yet the motioneye service refuses to start. I have all the dependencies installed correctly and motioneye is installed. pip, pip2, pip2.7 and pip3 all report that I have everything installed. python -V reads: Python 2.7.16.

> [root@localhost ~]# meyectl startserver -c /etc/motioneye/motioneye.conf 
Traceback (most recent call last):
  File "/usr/bin/meyectl", line 11, in <module>
    load_entry_point('motioneye==0.41', 'console_scripts', 'meyectl')()
  File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 254, in main
    import server
  File "/usr/lib/python2.7/site-packages/motioneye/server.py", line 31, in <module>
    import handlers
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 30, in <module>
    import config
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 32, in <module>
    import motionctl
  File "/usr/lib/python2.7/site-packages/motioneye/motionctl.py", line 28, in <module>
    import mediafiles
  File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 41, in <module>
    import uploadservices
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 27, in <module>
    import pycurl
ImportError: pycurl: libcurl link-time ssl backend (openssl) is different from compile-time ssl backend (none/other)

> [root@localhost ~]# systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Sun 2019-10-06 02:14:51 EDT; 11min ago
  Process: 19711 ExecStart=/usr/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 19711 (code=exited, status=1/FAILURE)

Oct 06 02:14:51 localhost.localdomain meyectl[19711]:     import motionctl
Oct 06 02:14:51 localhost.localdomain meyectl[19711]:   File "/usr/lib/python2.7/site-packages/motioneye/motionctl.py", line 28, in <module>
Oct 06 02:14:51 localhost.localdomain meyectl[19711]:     import mediafiles
Oct 06 02:14:51 localhost.localdomain meyectl[19711]:   File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 41, in <module>
Oct 06 02:14:51 localhost.localdomain meyectl[19711]:     import uploadservices
Oct 06 02:14:51 localhost.localdomain meyectl[19711]:   File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 27, in <module>
Oct 06 02:14:51 localhost.localdomain meyectl[19711]:     import pycurl
Oct 06 02:14:51 localhost.localdomain meyectl[19711]: ImportError: pycurl: libcurl link-time ssl backend (openssl) is different from compile-time ssl backend (none/other)
Oct 06 02:14:51 localhost.localdomain systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Oct 06 02:14:51 localhost.localdomain systemd[1]: motioneye.service: Failed with result 'exit-code'.

Additionally, [Step 7 in the guide](https://github.com/ccrisan/motioneye/wiki/Install-On-Fedora) says to make an init script, but does not describe how that is done (yes this may seem silly to some, but I'm just learning Linux and am not as familiar as others may be).

I'm out of ideas, will tackle this another day soon™

Mytherium, 2019-10-06:
The same thing happens to me on my Ubuntu 18.04.3 vm, so I'm not sure what I'm doing wrong

remisalmon, 2019-10-11:
Had the same issue on Fedora 30 and solved it with https://stackoverflow.com/questions/21096436/ssl-backend-error-when-using-openssl/21099222#21099222

(see https://github.com/ccrisan/motioneye/issues/1458)

Mytherium, 2019-10-12:
So I got 

> su -
> pip uninstall pycurl
> export PYCURL_SSL_LIBRARY=openssl
> pip install --compile pycurl

to work, the motioneye and motion services are running, however I'm not able to access the GUI from port 8765.

from the motioneye.log file:

> -- Reboot --
> Oct 12 00:47:44 localhost.localdomain systemd[1]: Started motionEye Server.
> Oct 12 00:47:45 localhost.localdomain meyectl[780]: Traceback (most recent call last):
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:   File "/usr/bin/meyectl", line 6, in <module>
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:     from pkg_resources import load_entry_point
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:   File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3191, in <module>
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:     @_call_aside
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:   File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3175, in _call_aside
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:     f(*args, **kwargs)
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:   File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3204, in _initialize_master_working_set
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:     working_set = WorkingSet._build_master()
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:   File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 583, in _build_master
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:     ws.require(__requires__)
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:   File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 900, in require
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:     needed = self.resolve(parse_requirements(requirements))
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:   File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 786, in resolve
> Oct 12 00:47:45 localhost.localdomain meyectl[780]:     raise DistributionNotFound(req, requirers)
> Oct 12 00:47:45 localhost.localdomain meyectl[780]: pkg_resources.DistributionNotFound: The 'pycurl' distribution was not found and is required by motioneye
> Oct 12 00:47:45 localhost.localdomain systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
> Oct 12 00:47:45 localhost.localdomain systemd[1]: motioneye.service: Failed with result 'exit-code'.
> Oct 12 00:51:30 localhost.localdomain systemd[1]: Started motionEye Server.
> Oct 12 00:51:30 localhost.localdomain meyectl[1255]:     INFO: hello! this is motionEye server 0.41
> Oct 12 00:51:30 localhost.localdomain meyectl[1255]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
> Oct 12 00:51:30 localhost.localdomain meyectl[1255]:     INFO: cleanup started
> Oct 12 00:51:30 localhost.localdomain meyectl[1255]:     INFO: wsswitch started
> Oct 12 00:51:30 localhost.localdomain meyectl[1255]:     INFO: tasks started
> Oct 12 00:51:30 localhost.localdomain meyectl[1255]:     INFO: mjpg client garbage collector started
> Oct 12 00:51:30 localhost.localdomain meyectl[1255]:     INFO: server started

Mytherium, 2019-10-13:
> [root@localhost extra]# netstat -tulpn | grep LISTEN
> tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      3292/sshd           
> tcp        0      0 192.168.2.66:8765       0.0.0.0:*               LISTEN      25705/python2       
> tcp        0      0 127.0.0.1:34185         0.0.0.0:*               LISTEN      2048/cockpit-bridge 
> tcp6       0      0 :::22                   :::*                    LISTEN      3292/sshd           
> tcp6       0      0 :::9090                 :::*                    LISTEN      1/systemd
> 

motioneye appears to be running successfully, and I'm able to modify the config file. However, something is preventing me from accessing the webgui at port 8765

JamBalaya56562, 2026-11-08:
Closing as answered above: `libcurl link-time ssl backend (openssl) is different from compile-time ssl backend` means pycurl had been built without the SSL backend of the system's libcurl, and rebuilding it with `PYCURL_SSL_LIBRARY=openssl`, as remisalmon linked (#1458), fixed it. This was the Python 2 install; current releases install pycurl with pip in a virtual environment, see the [README](https://github.com/motioneye-project/motioneye#installation).

closed 2026-11-08 by JamBalaya56562