motioneye-project/motioneye #1484: Motioneye on Rasberry pi - high CPU usage

seanocaster123, 2019-10-07:
Hi.

I running motion with motioneye on my pi4 with 1GB ram.

When i run nmon to monitor CPU usage i can see that 1 camera with a 1280x720 feed at 1 fps is using 110% of the CPU. Is that normal? If i add a 2nd camera it goes up to 220% and the pi starts really getting laggy.

takiainen, 2019-10-13:
You need to activate hardware acceleration, see here: https://github.com/ccrisan/motioneye/issues/930

lolren, 2019-10-20:
I have the same problem, also on pi4. i was wondering if there is an option to bypass motion all toghether? and use the url from the cameras directly to be encoded by ffmpeg to save them and also preview, without motion detection?

gusarg81, 2020-08-18:
Hi, I have similar problem on Pine Rock64 SBC (which has a RK3328 CPU). I've compiled FFmpeg (v4.3.1) with --enable-rkmpp (hardware encoding) and the CPU usage with Motioneye is extremely high, all four cores in 100%.

How do I know if Motioneye is using hardware encoding?

Thanks.

seanocaster123, 2020-08-19:
I couldn't get motion to work efficiently Eventually i ended up using zoneminder and it has been providing a reliable video stream of 1 camera for almost a year now. It only goes offline if the internet drops at the site, but otherwise, working perfectly

gusarg81, 2020-08-19:
> I couldn't get motion to work efficiently Eventually i ended up using zoneminder and it has been providing a reliable video stream of 1 camera for almost a year now. It only goes offline if the internet drops at the site, but otherwise, working perfectly

In my case Zoneminder used a lot of CPU as well, and its web interface is not functional at all (neither dynamic). Shinobi is another app, hard to make combinations of settings to work for a camera.

DdraigX, 2022-04-26:
Same issue here however I'm running motioneye in a docker container on a raspberry pi 3 b+

Does anyone know if recompling the ffmpeg/following the above guidance would help me? I'm new to docker so I figured I would ask.

zagrim, 2022-04-26:
I'm not completely sure which of the "above" you meant (perhaps h264_omx which is discussed in the linked issue, since h264_rkmpp to me looks not to be related to RPi). Anyhow, they all talk about h264, so if your camera happens to be using that encoding, then it might help somewhat. But it won't do miracles, since you might end up still using CPU for either encoding or decoding (unless I'm mistaken, h264_omx does only encoding, and h264_rkmpp seems to be a decoder). When doing motion detection (with saving of video) and streaming, both decoding and encoding are needed.

JamBalaya56562, 2026-11-10:
Closing, as this was answered: Motion decodes every frame to detect motion, which is the CPU load, and a hardware encoder for movies, a lower resolution or the camera's substream reduce it, as linked above (#930).