motioneye-project/motioneye #1486: Firefox Display Issue

bjs-pdx, 2019-10-08:
I have recently developed an issue with MotionEye when I try to switch a camera to full screen.  I'm using Firefox 69.0.2 but the issue also seems to occur with Google Chrome.  When I switch to full screen mode I get the attached screenshot.

/home/bjs/Downloads/Screenshot_2019-10-08 gumby(2).png

This started within the last few days.  
Running :
motionEye Version | 0.41
Motion Version | 4.0
OS Version | Ubuntu 18.04

I have other issues as well that started at the same time:
Multiple emails when camera goes offline using on_camera_lost
unusable video files (no recognized streams)

Really confused since, to the best of my knowledge, nothing changed on my setup.

bjs-pdx, 2019-10-08:
[image]

bjs-pdx, 2019-10-08:
There also seems to be intermittent failure to write movies:

[image]

JamBalaya56562, 2026-11-08:
Closing, as this got no replies, and it was with motionEye 0.41 and Motion 4.0 on Ubuntu 18.04, with several problems starting at once: the full-screen view, repeated `on_camera_lost` emails and unreadable movies. Please open a new issue for any of them that still happens with current releases, with the browser version or `motion.log`.

closed 2026-11-08 by JamBalaya56562