motioneye-project/motioneye #1492: Use SMB for saving files o

nishant2019, 2019-10-12:
MotioneyeOS has a option for SMB share but I canned find in motioneye with the complete OS. Can anyone assist me how a can use SMB option ?

Bromley47, 2020-05-24:
I have used the above instructions to save my Motioneye media on a WDMycloud NAS drive, due to limited storage on the SD card, plus the fact that I am using as a security camera and the camera could easily get pinched (along with the SD card and the saved images!) if someone realises what it is.

Had to re-figure some of the instructions (Googled them if they didn't work as should) and it saves the images to the MyCloud BUT only for while - at some stage it reverts back to saving them on the SD card, especially after a restart of the Pi. Any clues anyone - would be appreciated. To get it back to saving on the MyCloud drive I have to copy all the above again onto the Pi. I imagine I don't have to install all the above but not sure where to dive in and reinstall.

J-o-h-n-M, 2021-04-20:
Is this configuration still required? There is a GUI option "Enable Samba Server". But I cant get that to work. As the post here is from 2019, I was under the impression that it is now included in the GUI and no longer requires the separate installation of Samba.

starbasessd, 2021-04-20:
In motion/motionEye on another OS, you must still install SAMBA server as
part of that OS.
In motionEyeOS it is built in.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

J-o-h-n-M, 2021-04-20:
@starbasessd  thanks for the quick reply. I am running on RPI4 and see

```
motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606
```

So I would not need to configure anything seperate?

starbasessd, 2021-04-20:
Correct. You may want to head over to the motionEyeOS Github, too. More
pertinent info available.
https://github.com/ccrisan/motioneyeos/wiki

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-04-20:
Also, with a Pi4, the dev20201026 is much more stable.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

J-o-h-n-M, 2021-04-20:
Thanks! will have a look and also upgrade to 20201026.
Really appreciate the quick reply!

JamBalaya56562, 2026-11-08:
Closing, as this was answered: on Raspbian, motionEye does not set up Samba itself. To save to a share, either mount it in the OS (for example in `/etc/fstab`) and pick that folder under File Storage > Custom Path, or set `smb_shares true` in `motioneye.conf` and install `cifs-utils`, which adds Network Share there. "Enable Samba Server", which shares the media of the Pi itself, is part of motionEyeOS, which is maintained in its own repository (motioneye-project/motioneyeos).

closed 2026-11-08 by JamBalaya56562