motioneye-project/motioneye #1493: File format not recognized...

krreagan, 2019-10-12:
I installed MotionEyeOS on a Raspberry Pi 3B+ and everything seems to work as expected except when I try to play the files the apps say the file format is not recognized.
I've tried: Quicktime, VLC and Handbrake to play the files.
I've tried: (MPEG4) .avi, QuickTime (.mov) & H.264 (mp4)
All have the same complaint as file format not recognized (or similar wording).
I'm downloading to my Mac via the built in ftp server.
The video (live) displays well on my browser. Only when I try to play the saved files do I have issues.
Quality is set to default (75%)

njordan77, 2019-10-15:
I have a similar issue... maybe the developer can help? Do we need encoders installed seperately?

jasaw, 2019-10-16:
Which video encoder did you choose? Did you try the OMX variant and those without OMX?
Which version of MotionEyeOS are you running?

njordan77, 2019-10-16:
i tried them all - from MP4 to OMX MKV, HEC,....nothing works to be played. Ive seen others having similar issues and fixed by removing and re-entering the camera, did not work for me. The strange thing, files are generated, different file size in the range of a typical video file. but not playable at all.

I'm using the latest version (stable available of eyemotionOS)

jasaw, 2019-10-17:
Try a different video player?
Media Player Classic - Black Edition: https://sourceforge.net/projects/mpcbe/
Media Player Classic - Home Cinema: https://github.com/clsid2/mpc-hc/releases

krreagan, 2019-10-30:
I've tried QuickTime & VLC & Handbrake... nothing! All the formats seem corrupted or incomplete or the format is incorrect. They all report that the files are unrecognized or similar.
Latest version of MotionEyeOS

njordan77, 2019-10-30:
same here, tried so many different video player and formats - never ever had any similar incompatibility issues with media files. There must be something different wrong in some cases, as it seems to all people suffer the same issues. Is there anything we could try to get out of this "trap" :-) Thanks a lot Norbert

jasaw, 2019-10-30:
Have you guys tried playing the videos directly from a web browser? Chrome and Firefox?
Try running `ffprobe` on the video files and attach the output here.

njordan77, 2019-11-11:
i tried now also with the webbrowser, does not work any better. in general i guess VLC has one of the best codec supports, but it was worth a try.

ffprobe is not on the system. can you tell me how i could make it available? any package that needs to be installed?

njordan77, 2019-11-11:
if its allowed i could upload one of the smaller videos ~ 10MB

jasaw, 2019-11-12:
Please upload your broken video file somewhere like Dropbox or Google drive so we can download it to inspect.

njordan77, 2019-11-12:
Thanks....maybe the problem is something different in my case. I tried to prepare the dropbox upload in that i wanted to use samba to download a 40MB file to my Win10 machine. Copying picture files of 400k work fine...so athentication cannot be the problem....but the 40MB file starts very very slowly and breaks up after 1min at 5%....

Any idea why my SMB may have problems, i already turned on all SMB settings possible to ON.

jasaw, 2019-11-12:
@njordan77 do you have poor wifi connection? Try a different method to get the file out like scp?  If you run Windows, you could try winscp.

njordan77, 2019-11-15:
@jasaw no the wifi connection is ~80-90% RSSI, so that is definitely not a problem. I now downloaded with winscp and it works....but at 100kb/s...so very low, normally other Raspi machines provide almost the full bandwith of 802.11n. Any idea why this is the case? Ever seen any similar issues with MotioneyeOS. Due to this issue i tried to setup a complete new installation based in RPI 3B+, but does not help at all.

What i now found out, the files itself are OK, when downloading they can easily be played via VLC, BUT VLC via network seems to have the same problems as any other download procedure i tried....the very very low download kills everything...

Any suggestion is highly appreciated as this does not work very promising right now! Many many Thanks, Norbert

JamBalaya56562, 2026-11-08:
Closing, as the files were never checked with `ffprobe` or shared, and the later replies point to very slow transfers from the Pi, which can leave downloads incomplete. This was on motionEyeOS on a Pi 3B+, which is maintained in its own repository (motioneye-project/motioneyeos). Please open a new issue with the `ffprobe` output of a file if movies are still unplayable with a current release.

closed 2026-11-08 by JamBalaya56562