motioneye-project/motioneye #1494: after upgrade Debian 9 to 10 CURL_OPENSSL_3 error

bforpc, 2019-10-14:
Hi,

after upgradeing Debian strech to buster i get this error, when starting motioneye:
`
Oct 14 09:40:04 cb systemd[1]: Started motionEye Server.
Oct 14 09:40:04 cb meyectl[18924]: Traceback (most recent call last):
Oct 14 09:40:04 cb meyectl[18924]:   File "/usr/local/bin/meyectl", line 10, in <module>
Oct 14 09:40:04 cb meyectl[18924]:     sys.exit(main())
Oct 14 09:40:04 cb meyectl[18924]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 254, in main
Oct 14 09:40:04 cb meyectl[18924]:     import server
Oct 14 09:40:04 cb meyectl[18924]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 31, in <module>
Oct 14 09:40:04 cb meyectl[18924]:     import handlers
Oct 14 09:40:04 cb meyectl[18924]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 30, in <module>
Oct 14 09:40:04 cb meyectl[18924]:     import config
Oct 14 09:40:04 cb meyectl[18924]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 32, in <module>
Oct 14 09:40:04 cb meyectl[18924]:     import motionctl
Oct 14 09:40:04 cb meyectl[18924]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 28, in <module>
Oct 14 09:40:04 cb meyectl[18924]:     import mediafiles
Oct 14 09:40:04 cb meyectl[18924]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/mediafiles.py", line 41, in <module>
Oct 14 09:40:04 cb meyectl[18924]:     import uploadservices
Oct 14 09:40:04 cb meyectl[18924]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 27, in <module>
Oct 14 09:40:04 cb meyectl[18924]:     import pycurl
Oct 14 09:40:04 cb meyectl[18924]: ImportError: /usr/lib/x86_64-linux-gnu/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /usr/local/lib/python2.7/dist-packages/pycurl.so)
Oct 14 09:40:04 cb systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Oct 14 09:40:04 cb systemd[1]: motioneye.service: Failed with result 'exit-code'.

`
Clearing cache and reinstalling does not solve the problem

bfo.

bforpc, 2019-10-22:
i had try this with no success.

bfo

zagrim, 2019-10-24:
Have you also tried the solution given in #1346?

bforpc, 2019-10-24:
Very interesting:

```
pip uninstall pycurl
Uninstalling pycurl-7.43.0.3:
  Would remove:
    /usr/local/lib/python2.7/dist-packages/curl/*
    /usr/local/lib/python2.7/dist-packages/pycurl-7.43.0.3.dist-info/*
    /usr/local/lib/python2.7/dist-packages/pycurl.so
    /usr/local/share/doc/pycurl/AUTHORS
    /usr/local/share/doc/pycurl/COPYING-LGPL
    /usr/local/share/doc/pycurl/COPYING-MIT
    /usr/local/share/doc/pycurl/ChangeLog
    /usr/local/share/doc/pycurl/INSTALL.rst
    /usr/local/share/doc/pycurl/README.rst
    /usr/local/share/doc/pycurl/RELEASE-NOTES.rst
    /usr/local/share/doc/pycurl/examples/basicfirst.py
    /usr/local/share/doc/pycurl/examples/file_upload.py
    /usr/local/share/doc/pycurl/examples/linksys.py
    /usr/local/share/doc/pycurl/examples/opensocketexception.py
    /usr/local/share/doc/pycurl/examples/quickstart/file_upload_buffer.py
    /usr/local/share/doc/pycurl/examples/quickstart/file_upload_real.py
    /usr/local/share/doc/pycurl/examples/quickstart/file_upload_real_fancy.py
    /usr/local/share/doc/pycurl/examples/quickstart/follow_redirect.py
    /usr/local/share/doc/pycurl/examples/quickstart/form_post.py
    /usr/local/share/doc/pycurl/examples/quickstart/get.py
    /usr/local/share/doc/pycurl/examples/quickstart/get_python2.py
    /usr/local/share/doc/pycurl/examples/quickstart/get_python2_https.py
    /usr/local/share/doc/pycurl/examples/quickstart/get_python3.py
    /usr/local/share/doc/pycurl/examples/quickstart/get_python3_https.py
    /usr/local/share/doc/pycurl/examples/quickstart/put_buffer.py
    /usr/local/share/doc/pycurl/examples/quickstart/put_file.py
    /usr/local/share/doc/pycurl/examples/quickstart/response_headers.py
    /usr/local/share/doc/pycurl/examples/quickstart/response_info.py
    /usr/local/share/doc/pycurl/examples/quickstart/write_file.py
    /usr/local/share/doc/pycurl/examples/retriever-multi.py
    /usr/local/share/doc/pycurl/examples/retriever.py
    /usr/local/share/doc/pycurl/examples/sfquery.py
    /usr/local/share/doc/pycurl/examples/smtp.py
    /usr/local/share/doc/pycurl/examples/ssh_keyfunction.py
    /usr/local/share/doc/pycurl/examples/xmlrpc_curl.py
Proceed (y/n)? y
  Successfully uninstalled pycurl-7.43.0.3
:~ $ pip install pycurl
Requirement already satisfied: pycurl in /usr/lib/python2.7/dist-packages (7.43.0.2)

```

blackdot, 2019-11-11:
Same issue as bforpc. This forced pip to install pycurl again:
`pip install pycurl --upgrade`

amnesia106, 2019-11-18:
for me this workaround is not working. I uninstall pycurl, clean the cache and reinstall pycurl. the issue is the same. 
`motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-11-18 13:51:27 CET; 2s ago
  Process: 36267 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 36267 (code=exited, status=1/FAILURE)

Nov 18 13:51:27 smarthome meyectl[36267]:     import motionctl
Nov 18 13:51:27 smarthome meyectl[36267]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 28, in <module>
Nov 18 13:51:27 smarthome meyectl[36267]:     import mediafiles
Nov 18 13:51:27 smarthome meyectl[36267]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/mediafiles.py", line 41, in <module>
Nov 18 13:51:27 smarthome meyectl[36267]:     import uploadservices
Nov 18 13:51:27 smarthome meyectl[36267]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 27, in <module>
Nov 18 13:51:27 smarthome meyectl[36267]:     import pycurl
Nov 18 13:51:27 smarthome meyectl[36267]: ImportError: /usr/lib/x86_64-linux-gnu/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /usr/local/lib/python2.7/dist-pack
Nov 18 13:51:27 smarthome systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Nov 18 13:51:27 smarthome systemd[1]: motioneye.service: Failed with result 'exit-code'.`

gauravsharma013, 2019-12-23:
I'm facing similar issue and couldn't able to resolve it with uninstall/install pycurl.

fgr1986, 2020-05-14:
In my case, pycurl was not installed from pip.
If so, you can remove the folder and old lib with:
sudo mv /usr/local/lib/python2.7/dist-packages/pycurl* .

and then install it with
sudo apt install python-pycurl

closed 2022-03-19 by MichaIng