motioneye-project/motioneye #1496: No image nor video raspi 0 w

SSkull995, 2019-10-14:
Hello! 
I just installed the motion and motioneye plugin in my raspi 0 w, with a camera attached via the camera connector, not usb, but I cannot see images from it. It is weird because the camera is detected by my raspberry, if I run  vcgencmd get_camera I get **detected=1** and if I run **raspistill -o cam.jpg** I do not get errors, the cam.jpg file gets created. 
I runt also **raspistill -v -o test.jpg** and this is the log: 

raspistill Camera App v1.3.11

Width 2592, Height 1944, quality 85, filename test.jpg
Time delay 5000, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Camera component done
Encoder component done
Starting component connection stage
Connecting camera preview port to video render.
Connecting camera stills port to encoder input port
Opening output file test.jpg
Enabling encoder output port
Starting capture -1
Finished capture -1
Closing down
Close down completed, all components disconnected, disabled and destroyed

Test.jpg gets created as well. 

If I go to my ip:8765, the page shows up, I can select both MMAL and V4L2 Camera, however I don't get an image from both and the streaming ip, with :8081 doesn't work as well, it just shows "Safari cannot connect to the server". I also tried on Chrome, nothing. 

Any help will be much appreciated! Thanks!

SSkull995, 2019-10-14:
**LOG FILE BELOW**

-- Logs begin at Mon 2019-10-14 09:34:35 UTC, end at Mon 2019-10-14 10:20:00 UTC. --
Oct 14 09:34:44 PiGio systemd[1]: Starting motionEye Server...
Oct 14 09:34:44 PiGio systemd[1]: Started motionEye Server.
Oct 14 09:34:51 PiGio meyectl[375]: INFO: hello! this is motionEye server 0.41
Oct 14 09:34:53 PiGio motion[603]: [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Oct 14 09:34:54 PiGio motion[603]: [0] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
Oct 14 09:34:54 PiGio motion[603]: [0] [ALR] [ALL] conf_cmdparse: Unknown config option "camera"
Oct 14 09:34:54 PiGio motion[603]: [0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
Oct 14 09:34:54 PiGio motion[603]: [0] [NTC] [ALL] motion_startup: Logging to syslog
Oct 14 09:34:54 PiGio motion[603]: [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
Oct 14 09:34:54 PiGio motion[603]: [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Oct 14 09:34:54 PiGio motion[603]: [1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Oct 14 09:34:56 PiGio meyectl[375]: INFO: cleanup started
Oct 14 09:34:56 PiGio meyectl[375]: INFO: wsswitch started
Oct 14 09:34:56 PiGio meyectl[375]: INFO: tasks started
Oct 14 09:34:56 PiGio meyectl[375]: INFO: mjpg client garbage collector started
Oct 14 09:34:56 PiGio meyectl[375]: INFO: server started
Oct 14 09:34:57 PiGio meyectl[375]: ERROR: failed to disable motion detection for camera with id 1: Not Valid
Oct 14 09:35:39 PiGio systemd[1]: Stopping motionEye Server...
Oct 14 09:35:39 PiGio meyectl[375]: INFO: interrupt signal received, shutting down...
Oct 14 09:35:39 PiGio meyectl[375]: INFO: server stopped
Oct 14 09:35:39 PiGio meyectl[375]: INFO: tasks stopped
Oct 14 09:35:40 PiGio meyectl[375]: INFO: motion stopped
Oct 14 09:35:40 PiGio meyectl[375]: INFO: bye!
Oct 14 09:35:41 PiGio systemd[1]: Stopped motionEye Server.
Oct 14 09:48:37 PiGio systemd[1]: Starting motionEye Server...
Oct 14 09:48:37 PiGio systemd[1]: Started motionEye Server.
Oct 14 09:48:42 PiGio meyectl[15357]: INFO: hello! this is motionEye server 0.41
Oct 14 09:48:44 PiGio motion[15474]: [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Oct 14 09:48:44 PiGio motion[15474]: [0] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
Oct 14 09:48:44 PiGio motion[15474]: [0] [ALR] [ALL] conf_cmdparse: Unknown config option "camera"
Oct 14 09:48:44 PiGio motion[15474]: [0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
Oct 14 09:48:44 PiGio motion[15474]: [0] [NTC] [ALL] motion_startup: Logging to syslog
Oct 14 09:48:44 PiGio motion[15474]: [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
Oct 14 09:48:44 PiGio motion[15474]: [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Oct 14 09:48:44 PiGio motion[15474]: [1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Oct 14 09:48:46 PiGio meyectl[15357]: INFO: cleanup started
Oct 14 09:48:46 PiGio meyectl[15357]: INFO: wsswitch started
Oct 14 09:48:46 PiGio meyectl[15357]: INFO: tasks started
Oct 14 09:48:46 PiGio meyectl[15357]: INFO: mjpg client garbage collector started
Oct 14 09:48:46 PiGio meyectl[15357]: INFO: server started
Oct 14 09:48:46 PiGio meyectl[15357]: ERROR: failed to disable motion detection for camera with id 1: Not Valid
Oct 14 09:49:01 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:02 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:02 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:02 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:02 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:02 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:03 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:03 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:03 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:03 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:03 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:04 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:04 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:04 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:04 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:05 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:05 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:05 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:05 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:05 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:06 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:06 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:06 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
Oct 14 09:49:06 PiGio meyectl[15357]: WARNING: Connect error on fd 21: ECONNREFUSED
[... cut ...]
Oct 14 09:53:24 PiGio meyectl[15357]: ERROR: failed to disable motion detection for camera with id 1: Not Valid
Oct 14 09:53:24 PiGio meyectl[15357]: WARNING: Connect error on fd 17: ECONNREFUSED
Oct 14 09:53:24 PiGio meyectl[15357]: WARNING: Connect error on fd 17: ECONNREFUSED
Oct 14 09:53:24 PiGio meyectl[15357]: WARNING: Connect error on fd 17: ECONNREFUSED
Oct 14 09:53:24 PiGio meyectl[15357]: WARNING: Connect error on fd 17: ECONNREFUSED
Oct 14 09:53:25 PiGio meyectl[15357]: WARNING: Connect error on fd 17: ECONNREFUSED

zagrim, 2019-10-14:
You have incompatible versions of Motion and MotionEye. MotionEye 0.4x requires Motion 4.x, and you seem to have Motion 3.2.12. Please upgrade Motion (there are links and instructions on where to get it in the MotionEye Wiki under Installation in case the Linux ditribution you are using doesn't have more recent Motion in its repository).

SSkull995, 2019-10-14:
ok so I just upgrade motion, perfect will do it now. Thanks !

SSkull995, 2019-10-14:
Wait, one sec. I installed motion with apt install motion, shouldn't it get updated with sudo apt update/upgrade

zagrim, 2019-10-14:
Are you using Raspbian? Which version of it? I think only Raspbian Buster (most recent stable release) has recent enough Motion, for Stretch and older you should probably use the download link in the Wiki  Raspbian Installation instructions page.

SSkull995, 2019-10-14:
I just downloaded motion 4.1.x for jessie, because I cannot install MQTT on stretch, which I need as well

SSkull995, 2019-10-14:
**Update**
I just finished installing motion, now the steaming works and I get a grey picture with "unable to open video device"

SSkull995, 2019-10-14:
**New  Log**

-- Logs begin at Mon 2019-10-14 12:17:49 UTC, end at Mon 2019-10-14 12:23:41 UTC. --
Oct 14 12:17:58 PiGio systemd[1]: Starting motionEye Server...
Oct 14 12:17:58 PiGio systemd[1]: Started motionEye Server.
Oct 14 12:18:05 PiGio meyectl[376]: INFO: hello! this is motionEye server 0.41
Oct 14 12:18:08 PiGio motion[568]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Oct 14 12:18:08 PiGio motion[568]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Oct 14 12:18:08 PiGio motion[568]: [0:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "threshold_maximum"
Oct 14 12:18:08 PiGio motion[568]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
Oct 14 12:18:08 PiGio motion[568]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Oct 14 12:18:08 PiGio motion[568]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Oct 14 12:18:08 PiGio motion[568]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Oct 14 12:18:09 PiGio meyectl[376]: INFO: cleanup started
Oct 14 12:18:09 PiGio meyectl[376]: INFO: wsswitch started
Oct 14 12:18:10 PiGio meyectl[376]: INFO: tasks started
Oct 14 12:18:10 PiGio meyectl[376]: INFO: mjpg client garbage collector started
Oct 14 12:18:10 PiGio meyectl[376]: INFO: server started
Oct 14 12:18:27 PiGio motion[568]: [1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
Oct 14 12:18:27 PiGio motion[568]: [1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
Oct 14 12:18:27 PiGio motion[568]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
Oct 14 12:18:27 PiGio motion[568]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Oct 14 12:18:30 PiGio motion[568]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Oct 14 12:18:40 PiGio motion[568]: [1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
Oct 14 12:18:40 PiGio motion[568]: [1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
Oct 14 12:18:40 PiGio motion[568]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Oct 14 12:18:50 PiGio motion[568]: [1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled

zagrim, 2019-10-14:
Unfortunately I can't advice on solving the "vid_start: MMAL cam failed to open" error, hope someone more knowledgeable can help you with that.

SSkull995, 2019-10-14:
ok thanks! I just noticed that if I run vcgencmd get_camera I don't get any response, while just before updating motion yes

SSkull995, 2019-10-15:
Ok so I finished setting up the raspi again, formatting the sd and reinstalling everything. Weird thing, the video today was alive, I also streamed on Homekit with Homebridge. Now same issue, grey image and "cannot open video device". Plus, now I cannot stream via ip:8081

jasaw, 2019-10-16:
@SSkull995 Make sure your camera is connected properly to the Pi Zero W. Run `vcgencmd get_camera` again to make sure the GPU can detect the camera. If vcgencmd works, try raspistill.

SSkull995, 2019-10-16:
The i2c was the problem! I will let you know after a day of usage if the error will come back. Thanks!

Kind regards,
Giovanni

JamBalaya56562, 2026-10-29:
As found above, the camera could not be opened because of the i2c setup on the Pi, and the reporter got it working after fixing that. The setup here (MMAL, the legacy camera stack, motionEye 0.41) is no longer current; on current Raspberry Pi OS, the camera goes through libcamera, see https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack. Closing as completed.

closed 2026-10-29 by JamBalaya56562