motioneye-project/motioneye #1499: Unable to access MotionEye stream from the Internet

eWasteman, 2019-10-15:
I've had pretty good results with MotionEye thus far, so thank you for this awesome tool. Right now I've got a webcam plugged into a RasPi 3 successfully writing to my Google drive account. The last thing on my to-do list is getting the video stream accessible from the web, and then I've successfully bypassed all the mainstream home security camera companies 👍 

The issue I'm having is that my MotionEye video stream can only be accessed from within my LAN and not via some external connection. 

**From within my LAN, on any device, I can successfully access & view the camera by the following:**

http://192.168.0.63 _(DHCP reservation)_
http://192.168.0.63:8443
http://{MyPublicIPAddressHere}:8843

However, despite being able to access the stream (even via my public IP address) on a LAN-connected device. As soon as I try to access the stream externally (eg change my phone to 4g connection) it fails to connect.

Firstly, I'm definitely not ruling out this being a problem with my router and/or ISP as of right now. The reason I have used port 8843 both in MotionEye and in my Port-Forwarding rule on the router is because port 8843 for some reason is the only port I can prove can be reliably opened and closed on demand. I confirmed that by using one of the port checkers available online.

In the below issue, there is some suggestion that MotionEye (or Motion) may be some hidden setting that will not permit external streaming by default, but surely that cannot be true seeing as there is no mention of such thing on the MotionEye admin UI? 
https://github.com/ccrisan/motioneye/issues/888

Can anyone advise or recommend some troubleshooting tips to narrow down where the problem here might be?

Many thanks in advance

motionEye Version | 0.41
Motion Version | 4.2.2
OS Version | motionEyeOS 20190911

MikeMills727, 2019-10-27:
Not sure if this helps but I tried port forwarding and it worked great. The problem was no way to password protect it as far as I know. I was able to just type my main IP address with the port number. I have since deleted the forwarding rule. It was pretty fast with 4G Verizon network. Maybe somebody else will know how to port forward more securely?

Orwi53, 2019-11-02:
@MikeMills727 A safe way to do it is to remote-port-forward to a VPS server (forward a port of the VPS to your local port, where the web-interface is served), then, from anywhere the internet, local-port-forward from your pc/laptop to the VPS (forward a port of your laptop to the VPS's corresponding port that you set in the previous remote-port-forward step).
Then, open a browser on your laptop and, type http://localhost:port/

It works, although it's incredibly slow.

eWasteman, 2019-11-03:
Thanks for the suggestion, however I've just tried setting "authentication mode" to "disabled" and it hasn't made any difference to my issue unfortunately.

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye of 2019, and the last reply found no difference. Reaching the device from the internet depends on the router forwarding the port, which is outside motionEye. Since 0.44.0 the web UI uses login sessions, and for access from the internet a reverse proxy with HTTPS in front of it is the safer setup, see [Running Behind Nginx](https://github.com/motioneye-project/motioneye/wiki/Running-Behind-Nginx).

closed 2026-11-08 by JamBalaya56562