motioneye-project/motioneye #1504: On_event_end command sometimes does not stop

reparator, 2019-10-16:
Hi there,
I have a question regarding the "Run a Command" and "Run an End Command" in motioneye.
Here my versions:
motionEye Version 	0.39.3
Motion Version 	4.1.1
OS Version 	Raspbian 9.6
4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

MySystem is running for about a year now (on a RaspberryPi 3 B+) to surveil a birdhouse. To get sound recording via a microphone I added the following commands to /etc/motioneye/thread-1.conf:

`on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t; arecord -t wav -D plughw:1,0 /media/RaspiNAS/%Y-%m-%d/%H-%M-%S.wav`

This command starts a recording from the attached USB microphone when a motion is detected. The resulting wave-file is stored on a raspberryPi-NAS connected via cifs. The video-files go to the same storage device. Audio recording starts at the same time a video recording starts (nearly the same, because the date and time differ appr. 2 sec).

`on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t; pkill -SIGTERM -f '^arecord -t wav -D plughw:1,0 /media/RaspiNAS/'
`
This is the end command. It stops recording from the microphone by sendig the termination command.

All works most of the time but....

**1. The audio file is always about 30 to 50 sec. longer than the video file. Why that? Takes the pkill such a long time?
2. If I look to /var/log/motion.log I find something like this:**

```
pi@Meisenhaus:/etc/motioneye $ tail /var/log/motion.log -n30
[5072408:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[5072408:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
[5072408:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
[5072408:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[5072408:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
[5072408:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
[5072408:motion] [NTC] [ALL] config_camera: Processing camera config file thread-2.conf
[5072408:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
[5072408:motion] [NTC] [ALL] motion_startup: Logging to syslog
[5072408:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[5072408:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [WRN] [ENC] ffmpeg_set_codec: Preferred codec h264_omx has been blacklisted
2019-10-16 15:42:48: [sendmail]     INFO: sending email
2019-10-16 15:42:48: [sendmail]     INFO: email sent
[2:ml2] [WRN] [ENC] ffmpeg_set_codec: Preferred codec h264_omx has been blacklisted
Aufnahme: WAVE '/media/RaspiNAS/2019-10-16/16-01-32.wav' : Unsigned 8 bit, Rate: 8000 Hz, mono
Abbruch durch Signal Beendet ...
Aufnahme: WAVE '/media/RaspiNAS/2019-10-16/16-24-12.wav' : Unsigned 8 bit, Rate: 8000 Hz, mono
Abbruch durch Signal Beendet ...
Aufnahme: WAVE '/media/RaspiNAS/2019-10-16/18-10-46.wav' : Unsigned 8 bit, Rate: 8000 Hz, mono
```
**This last recording is not terminated and runs forever...**
If I look to the process list, the process of audio recording is still running even if the video recording has stopped a long time ago (about 4 hours later now):

```
pi@Meisenhaus:/etc/motioneye $ ps aux | grep arecord
root     13636  0.0  0.0   1900   372 ?        Ss   18:10   0:00 sh -c /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 2; arecord -t wav -D plughw:1,0 /media/RaspiNAS/2019-10-16/18-10-46.wav  &
root     13668  0.0  0.3   6232  2916 ?        S    18:10   0:12 arecord -t wav -D plughw:1,0 /media/RaspiNAS/2019-10-16/18-10-46.wav
pi       17438  0.0  0.0   4776   560 pts/0    R+   22:15   0:00 grep --color=auto arecord
pi@Meisenhaus:/etc/motioneye $ 
```
Has anyone an idea how I could localize the problem and fix it? **Why is the arecord-process sometimes not proper terminated?** The video recording stops as it should, the Email notification is send, but my audio recording did not get the sigterm-signal. Starting the recording works all the way right.

The problem is also, that if the process runs forever, the device is not free for another motion event to start...
Any help would be appreciated!
Regards Stefan

reparator, 2019-10-17:
Tried one more thing:
Using the pkill-command in a terminal as user pi, the operation is not permitted:
```
pi@Meisenhaus:/etc/motion $ pkill -SIGTERM -f '^arecord -t wav -D plughw:1,0 /media/RaspiNAS/'
pkill: killing pid 7121 failed: Die Operation ist nicht erlaubt
```
Any idea?

reparator, 2019-10-17:
And again:
Even a 
```
pi@Meisenhaus:/etc/motion $ sudo pkill 7121
pi@Meisenhaus:/etc/motion $ ps aux | grep arecord
root      7089  0.0  0.0   1900   404 ?        Ss   12:55   0:00 sh -c /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 2; arecord -t wav -D plughw:1,0 /media/RaspiNAS/2019-10-17/12-55-27.wav  &
root      7121  0.1  0.3   6232  2776 ?        S    12:55   0:25 arecord -t wav -D plughw:1,0 /media/RaspiNAS/2019-10-17/12-55-27.wav
pi       18783  0.0  0.0   4776   568 pts/0    S+   16:41   0:00 grep --color=auto arecord
pi@Meisenhaus:/etc/motion $ 
```
does not stop the process!
Only a 
`sudo kill -9 7121 `
kills the process.
Seems to be more Rasbian related than motioneye related, but anyway,
has someone an advice to fix this?

reparator, 2019-10-29:
Found out that it is related to the thread.conf used.
I added originally the arecord command to thread of an attached webcam (a Logitech C270 as V4L2 Camera). Here it does not workand the above described problems occur.
 If I use the other cam, an RaspberryPi Nightvision Cam as MMAL Camera, then everything works fine and the process stops as expected.
For me that helps but it is a little bit dissatisfying if you don know what the problem really is. It should work with the webcam either....

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye 0.39.3 on Raspbian 9, and you found a setup that works. Both commands run as the user that runs Motion, so `pkill` from the `pi` user is not allowed to stop a process started there, as you saw. Since 0.43.1 that user is the unprivileged `motion` user, and an end command running as the same user can stop what the start command began. Please open a new issue if it still fails with a current release.

closed 2026-11-08 by JamBalaya56562