motioneye-project/motioneye #1506: "Connection to Camera Lost", Gray Video

johnyradio, 2019-10-17:
i installed motioneye successfully with these steps.
https://github.com/ccrisan/motioneye/wiki/Install-On-Arch

camera:
- Device: /dev/video0 (wrapped)
- Card: USB Camera (093a:2620)
- Driver: gspca_pac7302
- Bus: usb-0000:00:1d.1-1

- browser: firefox
- the camera light is on.

i get the control panel, and all, at http://127.0.1.1:8765/
Streaming is on. 
i get a video-player at http://127.0.1.1:8081/

but the video is gray. starts out green

tuxslack, 2021-08-29:
Fabricante:  C3 Tech

Modulos carregados:

v4l2_common             4447  1 gspca_main
videodev              157398  3 gspca_main,gspca_pac7302,v4l2_common
i2c_core               48181  9 drm,i2c_dev,i2c_hid,i2c_piix4,drm_kms_helper,i2c_algo_bit,v4l2_common,radeon,videodev

$ lsusb
Bus 008 Device 034: ID 093a:2620 Pixart Imaging, Inc.

$ lsusb -v

Bus 008 Device 034: ID 093a:2620 Pixart Imaging, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x093a Pixart Imaging, Inc.
  idProduct          0x2620 
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          697
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           6
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
[... 818 lines cut ...]
        bFormatType             1 (FORMAT_TYPE_I)
        bNrChannels             1
        bSubframeSize           2
        bBitResolution         16
        bSamFreqType            1 Discrete
        tSamFreq[ 0]        16000
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x89  EP 9 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
        bRefresh                0
        bSynchAddress           0
        AudioControl Endpoint Descriptor:
          bLength                 7
          bDescriptorType        37
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x00
          bLockDelayUnits         0 Undefined
          wLockDelay              0 Undefined

starbasessd, 2021-08-29:
1) why 'hijack' a 2+ year old issue with no posts?
2) what is your language? I can't auto-translate.

closed 2021-08-29 by starbasessd