motioneye-project/motioneye #1507: [Feature Request] Prebuilt image for embedded x86 systems (#2)

Giga-Pudding, 2019-10-19:
Hi ccrisan,

I know [you have answered](https://github.com/ccrisan/motioneye/issues/619) this question already two years ago, but I would like to give two good reasons for providing a generic x86 image of Motion Eye.

The first thing is, that the currently supported hardware is (mostly) not powerfull enough, for the flawless processing of high resolution / high FPS Video. The second point is, that providing an x86 image would allow people to recycle old Hardware like Thin Clients and stuff.

I really prefer small single-board devices over fat old computers, but i think there are some good reasons though...

What do you think?

BR,
Giga

kleini, 2019-10-19:
The project already provides ready to use images for x86 hardware: https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
Therefore I do not really understand your feature request. Didn't you see yet these ready to use images?

Giga-Pudding, 2019-10-20:
Hi kleini,

to be honest, i wasn't aware that there is more available than the images listed [on this site](https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices). I searched for an x86 Image, and came across [this page](https://github.com/ccrisan/motioneye/issues/619).

I don't know much about docker and virtualization... these aren't images that can be flashed to an flashdrive and then be used to boot from, right?

kleini, 2019-10-20:
Then you need to file the request for the [motioneyeos](https://github.com/ccrisan/motioneyeos) project. This is the motioneye project covering only the motioneye software for itself and not any images shipping with the software.

closed 2019-10-20 by kleini