motioneye-project/motioneye #1511: After update to debian 10 python errors

bforpc, 2019-10-23:
After upgrading to debian 10 (from 9) i get these errors in syslog:

`
Oct 23 15:37:13 cb meyectl[656]: Traceback (most recent call last):
Oct 23 15:37:13 cb meyectl[656]:   File "/usr/local/bin/meyectl", line 10, in <module>
Oct 23 15:37:13 cb meyectl[656]:     sys.exit(main())
Oct 23 15:37:13 cb meyectl[656]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 254, in main
Oct 23 15:37:13 cb meyectl[656]:     import server
Oct 23 15:37:13 cb meyectl[656]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 31, in <module>
Oct 23 15:37:13 cb meyectl[656]:     import handlers
Oct 23 15:37:13 cb meyectl[656]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 30, in <module>
Oct 23 15:37:13 cb meyectl[656]:     import config
Oct 23 15:37:13 cb meyectl[656]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 32, in <module>
Oct 23 15:37:13 cb meyectl[656]:     import motionctl
Oct 23 15:37:13 cb meyectl[656]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 28, in <module>
Oct 23 15:37:13 cb meyectl[656]:     import mediafiles
Oct 23 15:37:13 cb meyectl[656]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/mediafiles.py", line 41, in <module>
Oct 23 15:37:13 cb meyectl[656]:     import uploadservices
Oct 23 15:37:13 cb meyectl[656]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 27, in <module>
Oct 23 15:37:13 cb meyectl[656]:     import pycurl
Oct 23 15:37:13 cb meyectl[656]: ImportError: /usr/lib/x86_64-linux-gnu/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /usr/local/lib/python2.7/dist-packages/pycurl.so)
Oct 23 15:37:13 cb systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Oct 23 15:37:13 cb systemd[1]: motioneye.service: Failed with result 'exit-code'.

`

zagrim, 2019-10-24:
It's unfortunate that you're still facing this, but opening new issues on it won't help solving it. Duplicate of #1494.

bforpc, 2019-10-24:
Yes, you right. But NOBODY is answering my question after now.

bfo

closed 2019-10-24 by bforpc