motioneye-project/motioneye #1515: High CPU load

ycodryn, 2019-10-25:
I`m using hassio and motioneye in a docker on a Synology DS718+ with Quad-core 2.3GHz and 8Gb Ram. This can transcode to 4K H.264/H.265. I have 6 ip cameras connected to eye but with view only (no record/motion detect) I have 90-100% cpu, with or without movie_passthrough on / ffmpeg_video_codec mp4:h264_vaapi. Using ffmpeg -encoders I have :

> ffmpeg version 4.1.4 Copyright (c) 2000-2019 the FFmpeg developers                                       
>   built with gcc 8.3.0 (Alpine 8.3.0)                                                                    
> --enable-vaapi --enable-vdpa
>  V..... h264_vaapi           H.264/AVC (VAAPI) (codec h264) 

so it could use acceleration but I don`t think that is using it. Anyway I`m also testing shinobi in docker and with full resolution, motion, record, my cpu is under 10%.... What could be the problem?

kleini, 2019-10-30:
A lot could be the problem. Hardware acceleration always depends on the available hardware and software support for hardware acceleration is very hard to implement.

jasaw, 2019-10-30:
@ycodryn motion has just added support for hardware accelerated decoder. See https://github.com/Motion-Project/motion/pull/1051 and https://github.com/ccrisan/motioneye/issues/1519
You have to find out which if your decoder is hardware accelerated from ffmpeg command.
If you don't select a hardware accelerated decoder as your "netcam decoder", motion will fall back to CPU decode, which most likely causes your high CPU load.

ycodryn, 2019-10-30:
@jasaw wow thanks. I`m gonna try it.

ycodryn, 2019-11-07:
@jasaw I may sound like a noob but how I`m gonna use it? I`m searching what to do but I see only that is implemented. I can use h264.vaapi.

OscarVanL, 2020-07-04:
For anyone else looking into trying to get hardware acceleration to work on a Synology NAS for any surveillance software, docker containers, or virtual machines, save your time and give up now.

I've spent weeks trying to get it to work. It seems Synology's OS just does not support it for third-party applications. Very few people have cracked it, most notably Plex have managed to barely get it to work, but even that requires a lot of tweaking and there are hundreds of posts where people struggle with it.

My conspiracy theory is that Synology do not want to make it easy to allow hardware acceleration for third-party surveillance camera software so you're required to buy the stupidly expensive £50-per-camera licensed software they make instead.

ycodryn, 2020-11-06:
Don`t think so. Zoneminder and Shinoby use does not have this problem.

closed 2020-11-16 by ycodryn

ycodryn, 2020-11-26:
>  Synology's OS

> For anyone else looking into trying to get hardware acceleration to work on a Synology NAS for any surveillance software, docker containers, or virtual machines, save your time and give up now. [...]

BTW maybe Synology's OS does not support this, MAYBE and I feel you brother in pain but using docker you are free of synology crap. The only problem is motion that is not using hwaccel.

rogierlommers, 2021-12-24:
How can I try this? My situation:

- Proxmox as server OS on the host
- A VM running Ubuntu
- In this VM, I run motioneye in a docker container

starbasessd, 2021-12-24:
With effectively 3 layers between motion/motionEye and the hardware, I doubt you can get hardware acceleration to work, period (doesn't matter what 3 layers, either). 
I can (and do, occasionally for testing) run VBox on Debian Host, with an Ubuntu or debian guest, running either raw motion/motionEye, or a motionEye docker container, without issue. 

Now, that said, I've never personally gotten proxmox to run any vm/container, ever.  
Not to say it can't be done, but my experience says there are issues.

rogierlommers, 2021-12-24:
Well, I hear what you say.
But I also have a question: how do you "create a stream" out of the RPI camera module?
I do it like this:

`/usr/bin/raspivid -o - -t 0 -hf -w 640 -h 360 -fps 25 | /usr/bin/cvlc -vvv stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554}' :demux=h264`

So I use raspivid to create the stream, which then goes to `cvlc` which in-turn creates a real rtsp stream. This stream is then "fetched" by motioneye which runs on my server (so not on the pi itself).

Can I do this more efficient? Any tips?

starbasessd, 2021-12-24:
@rogierlommers Are you running motionEye, motionEye in Docker, or what? Neither are very efficient, and you are running MANY layers with your system, there. 
What is the goal? What is the total environment?
The most efficient way to stream is to use streamEye on a Pi (what motioneEyeOS uses when switched to Fast Network Camera mode)
https://github.com/ccrisan/streameye
Other discussion:
https://github.com/ccrisan/motioneyeos/issues/2111
I have tested motionEyeOS dev20201026 FNC RTSP and H264 output to other motionEye hubs successfully. 
Trying to get a SBC (any SBC) to 'be all-end all' is not practical, depending on work load / appsw needed to run. Video processing is very CPU and GPU intensive. Networking / file processing is also.