motioneye-project/motioneye #1517: "Invalid configuration file line: motion_binary" when ever anything is changed

nraygun, 2019-10-29:
I get an "CRITICAL:root:failed to read settings from "/data/etc/motioneye.conf": invalid configuration file line: motion_binary" whenever I change any setting upon first boot of a freshly imaged SD card on my RPi3 with motioneyeos-raspberrypi3-20190911.img.
Any ideas?

nraygun, 2019-10-29:
To reproduce the problem, all I did was create a new microSD using the image above, turned on SSH amd hit apply. The system rebooted and I get the error above.

ccrisan, 2019-12-01:
Guys could you please try reproducing the issue with the *Expert Settings* section opened and see if the *Motion Binary* entry gets changed (by Chrome's autofill, for example) while you edit your options?

ccrisan, 2019-12-15:
Fixed via https://github.com/ccrisan/motioneye/commit/2caacf7f899070fc169ecdad5bc066214c1d1712.

Also related: https://github.com/ccrisan/motioneyeos/issues/2147.

closed 2019-12-15 by ccrisan