motioneye-project/motioneye #1518: Implement HomeKit

Saku241, 2019-10-29:
Hello 
Since the new iOS 13.2 update

> Home App

HomeKit Secure Video enables you to privately capture, store, and view encrypted video from your security cameras and features people, animal, and vehicle detection
HomeKit enabled routers put you in control of what your HomeKit accessories communicate with over the internet or in your home

It will be great to connect our camera with HomeKit 
Also I think a full step-by-step explaination could be necessary

Thanks !

Samfox2, 2019-10-29:
There is a homebridge-ffmpeg plugin for homebridge that allows homekit access for motioneye.

Saku241, 2019-10-29:
> There is a homebridge-ffmpeg plugin for homebridge that allows homekit access for motioneye.

I already tried but it doesnt work
Can you give me full step and Ill try to reproduce and describe what doesnt work

Samfox2, 2019-10-29:
You can install both packages via npm:
`npm install homebridge`
`npm install homebridge-ffmpeg`

Edit your config.json as described at the plugin pages and add something like this:
```javascript
{
            "platform": "Camera-ffmpeg",
            "cameras": [
                {
                    "name": "My camera",
                    "videoConfig": {
                        "source": "-re -i http://xxx.xxx.xxx.xxx:8081",
                        "maxStreams": 2,
                        "maxWidth": 1280,
                        "maxHeight": 720,
                        "maxFPS": 30
                    }
                }
            ]
        } 

```

Then start homebridge and add your cam to your home app. If itˋs working you can optimize stream quality with additional parameters as described at homebridge-ffmpeg project page.

Saku241, 2019-10-29:
It's actually working now, 
I got a preview of my camera in my HomeKit app but the stream is still a problem

I got this error code : 
`
[10/30/2019, 12:08:58 AM] [Camera-ffmpeg] Snapshot from Camera Surveillance at 480x270
[10/30/2019, 12:09:09 AM] [Camera-ffmpeg] Start streaming video from Camera Surveillance with 1280x720@299kBit
**[10/30/2019, 12:09:09 AM] [Camera-ffmpeg] ERROR: FFmpeg exited with code 1**
[10/30/2019, 12:09:24 AM] [Camera-ffmpeg] Start streaming video from Camera Surveillance with 1280x720@299kBit
**[10/30/2019, 12:09:24 AM] [Camera-ffmpeg] ERROR: FFmpeg exited with code 1**
`

Samfox2, 2019-10-30:
ffmpeg needs to transcode the stream, maybe your machine is too slow? If you are using a raspberry, omx decoder could be faster as it uses the GPU. Think you should post your problem to homebridge-ffmpeg issues page.

Saku241, 2019-10-30:
I’m using a basic authentification for my vidéo stream and I use this URL `http://user:password@192.168.1.x:8081`
Maybe this could be the cause ?

Saku241, 2019-10-30:
I fix it by refering the correct ffmpeg see my config.json now : 
```
{
	"bridge": {
		"name": "Homebridge",
		"username": "CC:22:3D:E3:CE:30",
		"port": 51826,
		"pin": "031-45-154"
	},

	"description": "This is an example configuration file with one fake accessory and one fake platform. You can use this as a template for creating your own configuration file containing devices you actually own.",
	"ports": {
		"start": 52100,
		"end": 52150,
		"comment": "This section is used to control the range of ports that separate accessory (like camera or television) should be bind to."
	},
	"platforms": [{
		"platform": "Camera-ffmpeg",
		### "videoProcessor": "/usr/bin/ffmpeg",
		"cameras": [{
			"name": "Camera Surveillance",
			"model": "MotionEye",
			"videoConfig": {
				"source": "-re -i http://user:password@192.168.1.2:8081",
				"vcodec": "h264_omx",
				"maxStreams": 2,
				"maxWidth": 1280,
				"maxHeight": 720,
				"maxFPS": 15
			}
		}]
	}]
}
```

ChrisLucas02, 2020-04-09:
Helle @Saku241,

I'm also trying to setup ip camera on homebridge. I can add the camera to the home bridge but I get the following error:
`
[4/9/2020, 1:57:17 PM] [Camera-ffmpeg] Snapshot from Camera One at 480x270
[4/9/2020, 1:57:17 PM] [Camera-ffmpeg] An error occurs while making snapshot request
[4/9/2020, 1:57:23 PM] [Camera-ffmpeg] Start streaming video from Camera One with 1280x720@299kBit
[4/9/2020, 1:57:23 PM] [Camera-ffmpeg] An error occurs while making stream request
[4/9/2020, 1:57:23 PM] [Camera-ffmpeg] ERROR: FFmpeg exited with code -2
`
I've installed homebridge-camera-ffmpeg plugin and config.json looks like this:
`
{
            "platform": "Camera-ffmpeg",
            "cameras": [
                {
                    "name": "Camera One",
                    "model": "MotionEye",
                    "videoConfig": {
                        "source": "-re -f mjpeg -i http://user:password@192.168.x.y:8081",
                        "maxStreams": 2,
                        "maxWidth": 1280,
                        "maxHeight": 720,
                        "maxFPS": 20
                    }
                }
            ]
        }
`
 I'm running a raspberry pi 3 on MotionEyeOS (camera). I've installed FFmpeg on the raspberry. When I use the source link in my browser, the camera feed appears. Did you configure anything special on your camera?

vmakwana, 2020-06-28:
After reading through this issue and https://github.com/homebridge-plugins/homebridge-camera-ffmpeg/issues/160, my solution was to install the ffmpeg, as it was missing on my setup (Raspberry Pi Zero W), to install,
`sudo apt-get -y install ffmpeg`

DanBurkhardt, 2020-10-28:
> Helle @Saku241,
> 
> 
> 
> I'm also trying to setup ip camera on homebridge. I can add the camera to the home bridge but I get the following error:
> 
> `[Camera-ffmpeg] Snapshot from Camera One at 480x270`
> 
> `[Camera-ffmpeg] An error occurs while making snapshot request`
> 
> `[Camera-ffmpeg] Start streaming video from Camera One with 1280x720@299kBit`
> 
> `[Camera-ffmpeg] An error occurs while making stream request`
> 
> `[Camera-ffmpeg] ERROR: FFmpeg exited with code -2`
> 
> 
> 
> I've installed homebridge-camera-ffmpeg plugin and config.json looks like this:
> 
> 
> 
> > 
> 
> 
> 
> {
> 
>             "platform": "Camera-ffmpeg",
> 
>             "cameras": [
> 
>                 {
> 
>                     "name": "Camera One",
> 
>                     "model": "MotionEye",
> 
>                     "videoConfig": {
> 
>                         "source": "-re -f mjpeg -i http://user:password@192.168.x.y:8081",
> 
>                         "maxStreams": 2,
> 
>                         "maxWidth": 1280,
> 
>                         "maxHeight": 720,
> 
>                         "maxFPS": 20
> 
>                     }
> 
>                 }
> 
>             ]
> 
>         }
> 
> 
> 
> > 
> 
> 
> 
>  I'm running a raspberry pi 3 on MotionEyeOS (camera). I've installed FFmpeg on the raspberry. When I use the source link in my browser, the camera feed appears. Did you configure anything special on your camera?

Jumping in here a little late, but has anyone considered the idea of using rtmp instead of FFMPEG? 

RTMP would let you set up your stream using existing configuration options￼￼￼, because you can already stream to rtmp.

Maybe I'm missing something here, let me know if this sounds workable.￼￼￼

JamBalaya56562, 2026-11-13:
As found above, HomeKit works through homebridge with homebridge-camera-ffmpeg, given motionEye's streaming URL as the source (and ffmpeg installed for the plugin). A built-in HomeKit service is not planned, so closing.

closed 2026-11-13 by JamBalaya56562