motioneye-project/motioneye #1521: Unable to Connect to Web Interface

fictionalname, 2019-10-30:
Hello,

I have installed motioneye on a raspberry pi zero, and it worked fine.

I then installed some other dependencies for running some Python scripts and now the web interface for motioneye is no longer available.  If I port scan on the device only port 22 is open.

Unfortunately, my Linux skills are not up to much so I am unable to resolve this issue.

I have tried going through the installation procedure again, and it says that the dependencies are installed and at the newest version.  I uninstalled and reinstalled motioneye and it doesn't give any errors.

Can you please help me to resolve this issue and get the web page working again.

Thanks!

zagrim, 2019-10-31:
Please provide the logs for MotionEye starting up:
`journalctl | grep -E "meyectl|motion"`

fictionalname, 2019-10-31:
Hello,

Thanks for responding.  The output is:

Oct 31 07:18:49 birdbox systemd[1]: Started motionEye Server.
Oct 31 07:18:50 birdbox motion[238]: Not starting motion daemon, disabled via /etc/default/motion ... (warning).
Oct 31 07:18:53 birdbox meyectl[260]: Traceback (most recent call last):
Oct 31 07:18:53 birdbox meyectl[260]:   File "/usr/local/bin/meyectl", line 6, in <module>
Oct 31 07:18:53 birdbox meyectl[260]:     from motioneye.meyectl import main
Oct 31 07:18:53 birdbox meyectl[260]:   File "/usr/local/lib/python3.7/dist-packages/motioneye/meyectl.py", line 28, in <module>
Oct 31 07:18:53 birdbox meyectl[260]:     import settings
Oct 31 07:18:53 birdbox meyectl[260]: ModuleNotFoundError: No module named 'settings'
Oct 31 07:18:53 birdbox systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Oct 31 07:18:53 birdbox systemd[1]: motioneye.service: Failed with result 'exit-code'.

fictionalname, 2019-10-31:
Hi,

With your help, pointing me at this log output, I seem to have found the problem.  Unfortunately, I'm unable to fix it, so any help would great!

It seems that I installed motioneye while the Python2.7 was the default, and thus it has placed all it's files in the Python2.7 folder.  I have now changed to Python3.7 which does not have the motioneye files - hence the errors in the log file.

I have tried to SUDO CP -a these folders into the Python3.7 folder in and attempt to fix the issue, but even with the SUDO, I'm getting an error saying :

cp: cannot create regular file '/usr/local/lib/python3.7/dist-packages/motioneye/templates/version.html': Permission denied

Is there any way I can either relocate these files to the right place, or reinstall motioneye into the Python3.7 folder?

Thanks for your help.

fictionalname, 2019-10-31:
Hmmm, I managed to copy the folders from the Python2.7 folder to the Python3.7 folder and it is clearly a more complex issue than I thought as it didn't resolve the issue.

I've tried restarting the service and even rebooting the Pi and I still can't connect and am still getting these issues : 

Oct 31 21:45:05 birdbox systemd[1]: Started motionEye Server.
Oct 31 21:45:08 birdbox motion[251]: Not starting motion daemon, disabled via /etc/default/motion ... (warning).
Oct 31 21:45:10 birdbox meyectl[247]: Traceback (most recent call last):
Oct 31 21:45:10 birdbox meyectl[247]:   File "/usr/local/bin/meyectl", line 6, in <module>
Oct 31 21:45:10 birdbox meyectl[247]:     from motioneye.meyectl import main
Oct 31 21:45:10 birdbox meyectl[247]:   File "/usr/local/lib/python3.7/dist-packages/motioneye/meyectl.py", line 28, in <module>
Oct 31 21:45:10 birdbox meyectl[247]:     import settings
Oct 31 21:45:10 birdbox meyectl[247]: ModuleNotFoundError: No module named 'settings'
Oct 31 21:45:10 birdbox systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Oct 31 21:45:10 birdbox systemd[1]: motioneye.service: Failed with result 'exit-code'.

zagrim, 2019-11-01:
MotionEye still requires Python 2.7, support for 3.x is not released yet.

fictionalname, 2019-11-01:
Ah ha!  That explains a lot then  Any idea if there is an update planned?

Thanks for your help.

closed 2019-11-01 by fictionalname

zagrim, 2019-11-02:
I don't know if there is any specific plan, but according to #1208 it would seem likely that the next bigger release 0.50 would run on Python 3.

foortizs, 2019-12-08:
I caused the same situation, where I definded python 3 as default on my raspberry pi, and as a result I got exactly the same error.
The solution was to force the motioneye service to use python 2.

In file: /etc/systemd/system/motioneye.service

I modified the following line:
ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

to:
ExecStart=**python2** /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

Now it works.

johnrego, 2021-07-09:
I get error when I run "systemctl start motioneye". However when I run "python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf" it works. Do you know how I can solve it? 

> I caused the same situation, where I definded python 3 as default on my raspberry pi, and as a result I got exactly the same error.
> The solution was to force the motioneye service to use python 2. [...]

starbasessd, 2021-07-09:
2 options:

1] change your default python to python2 (which will probably break other things)

2] edit file: nano /etc/systemd/system/motioneye.service
2] change line: 
2] ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
2] to:
2] ExecStart=python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

as mentioned above, as motionEye is still not certified for python3 and has many dependencies that do not translate from python2 to python3 easily.