motioneye-project/motioneye #1523: Motioneye Remote Camera, two issues.

SdeGeata, 2019-10-31:
Hello;

   I have two Raspberry Pi 3b+s that are both running MotionEye and Motion. Due to some CPU issues I have one set up with my Home Assistant automation and want to use that pi as the Hub to view an ip camera, as well as the piCam noir that is hardwired to the other pi.

   The first problem is that while I can view the remote camera from my hub, there are no "advanced settings" options and no "streaming URL" I also am unable to control the motion detection settings for the remote camera via the URL of the hub.

   That could be related to my second issue; the URL http://LocalHost:7999 doesn't even show the remote camera, although it shows up fine in the web interface (LocalHost:8765).

   Can someone help me find the error in my configuration???

SdeGeata, 2019-10-31:
It might be useful if I attached some screenshots huh?

The web interface of my "hub" - as you can see, both camera feeds:
[image]

The web control page of the same device - missing one feed:
[image]

The config for the remote camera:
[image]

More of the config - missing "streaming", among other things:
[image]

JamBalaya56562, 2026-11-10:
Closing, as both are how remote cameras work: a Remote motionEye Camera is run by the Motion on the other Pi, so the hub's own Motion control on port 7999 lists only the hub's local cameras, and motion detection for it is set and controlled on the other Pi. The hub also cannot tell that camera's MJPEG streaming URL, so it shows none; the other Pi's own motionEye shows it.

closed 2026-11-10 by JamBalaya56562