motioneye-project/motioneye #1529: Fullscreen auto-login

iRoboton, 2019-11-04:
Hi there,

First of all thanks a bunch for MotionEye, it looks awesome!

My current setup: RPi3B+ with MotionEyeOS installed, all good. Second RPi3B attached to my TV using HDMI. The second RPi starts the browser in kiosk mode, pointed to the MotionOS web interface.

Now, I need to login everytime the Pi reboots.. Idea is to launch in Kiosk mode, login as "User" to see the overview of connected camera's. So not 1 specific camera at a time, therefor the "streaming URL's are pointless, also, using these require manual login aswell. Can I append the login details to the URL? Tried

http://host/?username=X&password=X
http://username:password@host

No luck.
Thanks in advance

tjerkkamstra, 2021-06-11:
I have a similar question, but maybe a different solution: would it be possible to tell motioneye to consider requests from a specific IP as safe and auto login as admin?

starbasessd, 2021-06-11:
I just select "Remember Login" when logging in...

starbasessd, 2021-06-11:
If you have further questions, please open a new issue.

closed 2021-06-11 by starbasessd