motioneye-project/motioneye #1531: Ezviz C4s Dome Cam - grey screen Unable to open video device

denzelm5, 2019-11-06:
Good Day,

I have an issue with a using ezviz c4s camera on motion.(latest version on pi3 b) Im using rtsp://xxx.xxx.x.xxx
Type:Network Camera

[image]

[image]

Please advise

Regards

iRoboton, 2019-11-06:
Try selecting UDP as source...
When you add new camera, below the URL, there is a field "Camera". This defaults to "RTSP/TCP camera", select the UDP one in the dropdown and try again.

denzelm5, 2019-11-06:
Hi iRoboton,

I selected the UDP option but I still have the grey screen.

Regards

RDamman, 2020-05-16:
I also have an Ezviz C4S. But i can only get pictures of it via:
**http://admin:XXXXXX@192.168.x.x/Streaming/channels/1/picture**
where XXXXXX is the verification code on the backside of the camera.
You can see that the camera supports rtsp by using the link:
**http://admin:XXXXXX@192.168.x.x/Streaming/channels/1/** or even **http://admin:XXXXXX@192.168.x.x/Streaming/channels/** for more capabilities.

Opening the rtsp link **rtsp://admin:XXXXXX@192.168.X.X:554/Streaming/Channels/1** in VLC gives a popup where user **admin** and the verfication code doesn't seem to be valid. Hopefully someone knows the password. Or maybe the verification code is only valid when the camera isn't paired with the Ezviz software / app.

kathgironpe, 2020-06-16:
Selecting UDP worked for me.

bilbolodz, 2021-01-11:
@RDamman have you managed to get rtsp stream using vlc? I've the same problem. ONVIF Device Manager is able to display video stream (shows rtsp://IP_CAMERA:554/Streaming/Channels/1?transportmode=unicast&profile=Profile_1" using credentials: admin/verification code.

RDamman, 2021-01-11:
No. But I think the demuxer must be set manually instead of automatically in VLC. This can also be done on the command line.

denzelm5, 2021-01-11:
Hi,

I did get the issue fixed, I had to use firmware V5.3.1 build 180109 for the ezviz C4S.

Regards

bilbolodz, 2021-01-11:
Could you please share these firmware? I've sent an email to ezvis support but never got an answer.

RDamman, 2021-01-11:
> 
> I did get the issue fixed, I had to use firmware V5.3.1 build 180109 for the ezviz C4S.
> 
Please, do you want to share this firmware?

denzelm5, 2021-01-11:
How does one upload a file? :(

RDamman, 2021-01-11:
For large files you can use wetransfer.com. You can use my email address (damman at wxs.nl) or please mail it to me directly.

denzelm5, 2021-01-11:
Sent via email
@bilbolodz send me your email address I can pop it to you.

bilbolodz, 2021-01-11:
Email sent

RDamman, 2021-01-11:
The following information is based on the directions of Denzelm5:

The firmware file (ezviz.dav) can be applied in this screen of Ezviz Studio:
[image]

If you don't have access to this (options) you have to alter the configuration file of Ezviz Studio. So:
- Install the Ezvis Studio software on a windows machine.
- Create an account etc.
- C:\Program Files (x86)\Ezviz Studio\config
You looking for the AppConfig file.
- Make a copy of this file just incase things go south.
- Add these lines to the file:
[LocalOperation]
Show=1
- Save the file and paste it the original location.
- Restart Ezviz Studio when already running.

RDamman, 2021-01-11:
Yes, it works!!!! Download link to file [https://1drv.ms/u/s!AnUCD5VhpT5umQMTuhAhjYKmZhCN?e=MAJOKU](url). Now VLC can open the stream rtsp://admin:XXXXXX@192.168.X.X:554/Streaming/Channels/1.

bilbolodz, 2021-01-11:
It's working THANK you. I've successfully flashed my camera with new firmware and now RTSP is working fine.

RDamman, 2021-01-12:
There is a disadvantage: The webserver on the camera (nginx) seems not to be running anymore. It's impossible to download a snapshot photo or retrieve information about the streams, as described above, as with the old firmware.

denzelm5, 2021-01-12:
Hi,

Nice to see there is some progress. Can I ask what Is the desired result you guys trying to achieve ? Do you want to do AI? integrate into homeautomation ?

bilbolodz, 2021-01-12:
Sorry (for that) but I will add my camera to Zomeminder ;-)

JamBalaya56562, 2026-11-10:
Closing, as this was solved on the camera's side: with Ezviz firmware V5.3.1 build 180109 the C4S serves RTSP at `rtsp://admin:<verification code>@<ip>:554/Streaming/Channels/1`, as the reporter and others confirmed.

closed 2026-11-10 by JamBalaya56562