motioneye-project/motioneye #1538: Motioneye don't start after installation on Raspbian Buster

Piteglio, 2019-11-11:
Hi,
I tried and tried to install, three times, motioneye on raspbian buster following the directions of the page
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian.
I installed everything with precision but at the end motioneye does not start.
I need a little help to figure out where the problem is.

jasaw, 2019-11-11:
Try replacing pip with pip2 in the instructions.

Piteglio, 2019-11-14:
Hi, 
after a sleepless night, I made it.
A problem arose: everything works but the color of the camera pi is all carried on the red hue.
The funny thing is that at night, with the garden lights on, the colors are natural: the lawn is green and so on. 
What can be the problem?

zagrim, 2019-11-15:
Please see #575, even though it talks about green tint instead of red you might be able to adjust the automatic white-balance by putting similar `mmalcam_control_params` line to "Extra Motion parameters" field in MotionEye.
Please see https://motion-project.github.io/motion_config.html#mmalcam_control_params to pointers to more detailed info on the available options (I've not used that so can't be more specific).

Piteglio, 2019-11-16:
Hi, 
I followed suit and tried via ssh to install sudo modprobe bcm2835-v4l2.
Things have not changed.
Raspberry pi 3 b+ and I followed the directions to the letter Install On Raspbian, before.
[image]
In the evening with the artificial lights the camera pi provides almost natural colors, as you see.
In the rest of the day, with natural light, there is a single color that gives a faintly reddish shade.

amnesia106, 2019-11-20:
> Hi,
> after a sleepless night, I made it.

what was the issue?

hexordecimal, 2019-11-20:
do you have auto brightness set on your motioneye settings in the web ui? what is your exposure settings for your v4l2? have you tried looking for information on v4l2-ctl to control the video settings for your pi camera?[ v4l2 control commands](https://www.mankier.com/1/v4l2-ctl)

closed 2021-03-05 by starbasessd