motioneye-project/motioneye #1539: Proposal: If the device is in the network then recording is disable (vice versa)

girardpeter, 2019-11-12:
I hope this option will be possible for the next update.

Explanation:
If the mobile phone device is connected in the wifi network then motion eye will stop/disable the recording.

If the mobile phone device is not connected in the wifi network then motion eye will activate/enable recording if there is motion detected.

girardpeter, 2019-12-05:
also if the bluetooth from mobile phone is within the range of raspberry pi. the motion eye will not save motion. vice versa

girardpeter, 2020-01-05:
Hi Cris,
Hope this will be available soon.

klausmcm, 2020-04-09:
I'd love to see this feature too. Phone would act as an indicator of whether someone is at home or now. Could maybe provide an option to ping a device name/IP every so often?

klausmcm, 2020-04-09:
Could do the symlink approach in #659 in the meantime.

closed 2021-03-05 by starbasessd