motioneye-project/motioneye #1540: Recorded videos always stutter after a few seconds

der-gt, 2019-11-13:
Hello,

I have installed Motion on a pi4 4gb running raspbian via usb3.0 on ssd.
So I think hardware can't be the bottleneck here.

Everytime Motion detects something and records a video (via h264/omx), at first I got flawless playback but after about 3 seconds the video go all stuttering.
I tried #930 but it didn't help with my problem.

Can anyone help me out? This gets annoying, went from pi zero to pi3b+ to pi4 but still no flawless recording...

Heres a sample video which I recorded, if someone needs other files or logs, please let me know.
https://filebin.net/oy6t2wkew4sbe6gm/Cam_00-16-55.mp4?t=uk0avkqn

Thanks, der-gt

jasaw, 2019-11-14:
I don't know what is causing your stuttering video. I have a few suspicions.
1. Frame rate dropped when there are big changes between frames, e.g. the person walks nearer to the camera. What kind of camera is it? Is the camera communicating via wifi or ethernet or connect directly to the Pi4 camera port? Try waving hand at the lower right corner of the camera view where you get smooth playback.
2. Frame rate dropped because camera auto-brightness captures slower when there's less light, e.g. person walks past a light source, making the scene darker. Are you able to test your set up in a well lit environment?
3. Did you set up "on motion detect" actions? Running something CPU or IO intensive may cause frame rate to drop.
4. Could also be a Motion software issue. If all else fails, you might want to report this issue to motion github project.

der-gt, 2019-11-14:
Hallo and thanks for your reply.
1.) I think I can exclude this. Because when you compare other recorded videos, you see this happens always after 3secs, even if there is only lighlty movement an so no big frame changes.
Camera is connected via wifi.
2.) This I can exclude, too. If you are in IR Nightmode the same is happening.
3.) Yes I set several actions, uploading files via FTP and sending pics via eMail. But I think the Pi4 has more than enough power?
4.) I thought, that I've done that with this thread here? Am I false here?

Thanks and best regards.

der-gt, 2019-11-14:
so I can clear I out that it has to be a software issue inside motion. with shinobi I got recording without any stuttering...

Sean-Arbology, 2020-07-10:
Hi, I *think* I am having the same trouble.

I have done a fair amount of A/B testing with motioneyeos, and motioneye running on DietPi.
Rpi/mmal camera straight into the board.

Motioneyeos will run on Pi3 A+ with camera at 800x600 @ 30fps, including motion detection and recording video.

Motioneye on DietPi comes with an older version of motion (4.1.1) to begin with. This would not stream above 10-15fps, and video recording was painfully slow. Updating to motion 4.2/4.3 (I tested both) fixed the streaming/motion issue. So to be clear it is *very* close to being perfect - with Motioneye running on Dietpi and upgraded motion version to 4.2/4.3 I can stream and detect motion at 800x600 30fps.

Where the process is still falling over is when recording, and this is where I am a bit baffled as to how I can proceed.
I have checked that my ffmpeg includes the x264_omx codec - it does. So I am not sure that #930 applies?
My ffmpeg version is 4.1 which *could* be the issue? I have tried #930 as well, but compiling ffmpeg fails repeatedly and takes a seriously long time, so I am reluctant to keep bashing away at that for now unless I don't get an answer here.

**Some things I noticed**

- Watching HTOP - motioneyeos seems to have 4 instances of motion running at all times?

- When motion event is triggered in motioneyeos, the python script to launch the relay event triggers, and the stream stays at 25-30fps (steady/good) throughout. After approx 10 seconds AFTER the motion event, ffmpeg launches to do the write. Video stays smooth throughout, and the recorded video is free from jitter or any issues.

- When motion event is triggered in motioneye on dietpi, the python script triggers and ffmpeg triggers at the same time, video is clean for approx 2 seconds then goes jittery. This is carried into the recording (i.e. is not just an artefact of the video/stream).

Attached are two screenshots - one of htop showing on dietpi&motioneye (single motion process), and one of motioneyeos (4x motion processed)

[image]
[image]

Looking at the ffmpeg command between the two, as well as all other configuration (camera-1.conf, motion.conf, motioneye.conf) they are identical. But clearly something is different here. The only obvious difference to me is the ffmpeg version. Is it likely to be as simple as that?

If only there was a debian package for ffmpeg4.2!

rdmueller, 2020-08-05:
hm. I've got exact the same problem with different raspberry pit devices and the raspberry pi camera (no network).

closed 2021-03-05 by starbasessd