motioneye-project/motioneye #1541: Security risk for your Google Drive account

ygreq, 2019-11-13:
Hi there!

I hope nobody else wrote about this. Otherwise please provide link. Thank you

I got a warning from Google saying that motionEye <<has access to: See, edit, create, and delete ALL of your Google Drive files.>> The emphasis is mine and it is the problem I want to address. I might be wrong, but generally an app has access to only a part of Google Drive, mainly the folder that it creates, and not all of Google Drive. And I find this situation to be a big security risk. 

Is there something I should do so it has access only to the folder it creates? Is this normal behaviour for apps that have access to GDrive so my worry is not founded? 

Thank you for looking into it!

ccrisan, 2019-11-17:
Also related https://github.com/ccrisan/motioneyeos/issues/2191.

ccrisan, 2019-11-29:
See https://github.com/ccrisan/motioneyeos/issues/2191 for a detailed explanation and current status.

ccrisan, 2019-12-16:
Should be working fine now.

closed 2019-12-16 by ccrisan