motioneye-project/motioneye #1542: Installation process on raspbian

davidcgu, 2019-11-15:
Good morning

I'm following the installation steps for raspbian with latest raspbian image downloaded today, I'm fine until reaching the step as follows;

 sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf

And get:

cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory

I remeber to follow the process one year ago and was working so I assume something has changed on latest versions, any idea?

Thanks in advance

zagrim, 2019-11-15:
Searching issues with the error suggest that you might have not run `sudo pip install motioneye` with `sudo`, in that case the stuff would probably go to `.local/share/motioneye/extra` instead of `/usr/local/share/motioneye/extra`.

davidcgu, 2019-11-15:
I should be a moron to not realise that..... sorry

It works, thanks a lot for your help!

zagrim, 2019-11-16:
Ok, great! It would be nice if you also closed this issue now that's resolved.

closed 2019-11-16 by davidcgu