motioneye-project/motioneye #1544: Motioneye freezes with "mjpg client timed out" every few days

Dinth, 2019-11-15:
Im running Motioneye official docker container.
Normally it works perfectly fine with 4-5 network cameras but every few days Motioneye all streams break down. Some cameras are showing image frozen several hours ago and some streams are completely offline.
Logs show
```
   ERROR: mjpg client timed out receiving data for camera 3 on port 8083
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081
```

wzbart, 2020-01-22:
> Im running Motioneye official docker container.
> Normally it works perfectly fine with 4-5 network cameras but every few days Motioneye all streams break down. Some cameras are showing image frozen [...]

Have you found the solution? I'm running in to very same issue.

Dinth, 2020-01-22:
No i havent so far. I have removed a couple of cameras from motion eye leaving just three. The situation has improved slightly.
It would be nice to have some kind of watchdog build into motioneye which checks if ffmpeg processes are working.

Legolaszstudio, 2020-08-02:
It's funny how it hasn't been fixed **yet**.....

MichielvanBeers, 2020-08-03:
Agreed with @Dinth. Is there anyway how we can for example make our own script that checks the motioneye log for this error? Since motioneye is aware that the issue takes place, we should be able to restart the service, right?

STUKguy, 2020-12-25:
Did anyone find answer for this?

starbasessd, 2020-12-25:
Are you getting it? I have run docker on a VM, but never had it crash. What version of motion & motionEye & OS is showing in the docker motionEye WebGUI?

MichielvanBeers, 2020-12-27:
Not sure if that question was directed at me, but I currently have it running stable by triggering a cronjob every night that reboots the device. I run the latest version: 

motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | motionEyeOS 20200606

Woodpeckercz, 2020-12-28:
> Not sure if that question was directed at me, but I currently have it running stable by triggering a cronjob every night that reboots the device. I run the latest version:
> 
> motionEye Version | 0.42.1
> Motion Version | 4.2.2
> OS Version | motionEyeOS 20200606

I don't know how to create cron jobs yet, would you mind sharing your code for restarting the camera every night ?
Seems like my camera looses connection every 24-48h.

MichielvanBeers, 2020-12-28:
Yeah no problem:

Make sure that SSH is turned on in the Service settings. Login via SSH on your motioneye server with root and the password that you've set for the admin user in the webinterface. 

After logged in, type `crontab -e` and paste the following line in the crontab page: `0 4 * * * reboot now >/dev/null 2>&1` . Press `ctrl + X` and save the changes. Your device should now reboot everyday at 4:00 am if all is well.

Woodpeckercz, 2021-09-09:
> Yeah no problem:
>
> Make sure that SSH is turned on in the Service settings. Login via SSH on your motioneye server with root and the password that you've set for the admin user in the webinterface. [...]

Thank you so much.

leprechaun17, 2023-08-13:
> crontab -e

How do you perform this when motioneye is installed as a docker container on unraid?

MichielvanBeers, 2023-08-14:
@leprechaun17 you should run the command inside of the Docker container. I don't use Unraid but the command line and then it would go like [this](https://www.digitalocean.com/community/tutorials/how-to-use-docker-exec-to-run-commands-in-a-docker-container). Keep in mind that Docker containers are stateless and that restarting it, will remove the changes you've made. You can find some Unraid specific info [here](https://forums.unraid.net/topic/133840-docker-commands-inside-a-container/).

JamBalaya56562, 2026-11-12:
This was reported on 0.42.1 and motionEyeOS 20200606, and motionEye's MJPEG client has changed a lot since, e.g. in how it requests Motion's stream (#2432) and keeps the connection (#3266). As there were no reports from current versions, I'm closing this. If streams still freeze with "mjpg client timed out" on the current version, please open a new issue with the motionEye and Motion logs around the freeze.

closed 2026-11-12 by JamBalaya56562