motioneye-project/motioneye #1547: curl on webcontrol to add text only works without any spaces

discarn8, 2019-11-17:
I'm trying to use the webcontrol portion of motioneyeos and while it works  - it only works when I don't use any spaces.  I.e:

cputemp=$(awk '{print $1/1000 * 1.8 + 32}' /sys/class/thermal/thermal_zone0/temp | awk '{printf("%d\n",$1 + 0.5)}')

and then
$ curl http://localhost:7999/1/config/set?text_left="Driveway__$cputemp"
works and displays:    Driveway__110
which is fine but annoying with the underscores

But...
curl http://localhost:7999/1/config/set?text_left="Driveway      $cputemp"
fails with:   Driveway

Has anyone else noticed this and if so, know how to get around it?

Raspberry Pi 2
motionEye Version | 0.39.2
Motion Version | 4.1.1
OS Version | motionEyeOS dev20180712

$ cat /data/etc/motion.conf
-------------------------------------------------------------------------
# @normal_password 
# @enabled on
# @admin_username admin
# @show_advanced on
# @admin_password 0000000000000000000000000000000000000000
# @normal_username user

setup_mode off
webcontrol_localhost off
webcontrol_html_output on
webcontrol_port 7999
webcontrol_parms 1
thread thread-1.conf

zagrim, 2019-11-17:
This is just guessing, but does it make any difference if you enclose the whole command in quotes (e.g. `"curl http://localhost:7999/1/config/set?text_left=Driveway $cputemp"`? How about doing URL encoding for the space: `"curl http://localhost:7999/1/config/set?text_left=Driveway%20$cputemp"` ?

discarn8, 2019-11-18:
zagrim - you're a genius and a scholar and a gentleman - that nailed it.  Thank you!

curl http://localhost:7999/1/config/set?text_left="DRIVEWAY%20%20$cputemp%46"

DRIVEWAY  113F

Success!
[Now... I need to figure out how to get the degree symbol to work]

zagrim, 2019-11-18:
Glad it worked. If you need to easily get properly encoded URLs (or parts of the URL like here), there are online converters available, like urlencode.org.

discarn8, 2019-11-19:
Zagrim - Thanks for that idea.  I tried doing the urlencode of the degree symbol ° [%C2%B0] to get "DRIVEWAY  113°F" and pasted it into the curl command, like so: 
curl http://localhost:7999/1/config/set?text_left="DRIVEWAY%20%20$cputemp%C2%B0%46"

....but it always comes out as "02November0" on the camera text overlay, for some reason.

Oh well - it's working sufficiently for me, as is - I'll not waste too much more time on aesthetics.

Thanks again!

closed 2019-11-19 by discarn8