motioneye-project/motioneye #1548: I want control PTZ through physical buttons/joystick/gamepad?

KjellKriminell6, 2019-11-18:
Hi! I would like to control my PTZ ip camera through motioneye action buttons physically. Pan , tilt, zoom, record.
Is that possible through some joystick shield or USB-gamepad?

hexordecimal, 2019-11-20:
python has[ joystick](https://pypi.org/project/joystick/) and heres this https://github.com/ccrisan/motioneyeos/pull/2016, pythons a pretty easy language

hexordecimal, 2019-11-26:
ok so i figured something out, if you have a usb gamepad. this only works if you already have scripts that work as action buttons.
1.plug it in and check /dev/input for a event0.
2.press a button and then cat /dev/input/event0( should see unreadable data)
3.`sudo pip install evdev`
```
import os
from evdev import InputDevice, categorize, ecodes, KeyEvent
gamepad = InputDevice('/dev/input/event0')
 
for event in gamepad.read_loop():
    if event.type == ecodes.EV_KEY:
        keyevent = categorize(event)
        if keyevent.keystate == KeyEvent.key_down:
            if keyevent.keycode[0] == 'BTN_A':
                #place shell to move up os.system(sh up.sh)
                #place curl command to move up os.system(curl -x 192.168.1.2)
                print "UP"
            elif keyevent.keycode == 'BTN_Y':
                #place shell to move down os.system(sh up.sh)
                #place curl command to move down os.system(curl -x 192.168.1.2)
                print "DOWN"
            elif keyevent.keycode == 'BTN_B':
                #place shell to move left os.system(sh up.sh)
                #place curl command to move left os.system(curl -x 192.168.1.2)
                print "LEFT"
            elif keyevent.keycode == 'BTN_X':
                #place shell to move right os.system(sh up.sh)
                #place curl command to move right os.system(curl -x 192.168.1.2)
                print "RIGHT"
```

closed 2021-03-05 by starbasessd