motioneye-project/motioneye #1549: motion_control_localhost Setting Not Honored

ryancasler, 2019-11-18:
After setting motion_control_localhost to false, this is all that I get when I navigate to `http://camera.ip:7999`

[image]

This is running on the following config:
motionEye Version 0.41
Motion Version 4.2.2
OS Version Raspbian 10

This was working in previous versions.  Did something change in the latest to make this not work anymore?  It is working in:
motionEye Version 0.41rc1
Motion Version 4.1.1
OS Version Raspbian 9.11

Could it be the version of motion?  Is there any way to downgrade to the older version?

zagrim, 2019-11-19:
Please try with `webcontrol_localhost` (maybe you also need to set `webcontrol_parms`). Motion changed some of the config parameters (and it should also warn about those in the logs if you look in there). 
https://motion-project.github.io/motion_config.html#OptDetail_Webcontrol

ryancasler, 2019-11-19:
Ah ha!!  That explains why things looked so different.  What I was actually seeing was the "minimal" text/only configuration.  Had to set webcontrol_interface to 2 to get the old legacy page to load.  Thanks!!

closed 2019-11-19 by ryancasler