motioneye-project/motioneye #1552: 0% GPU usage

Maxloader2000, 2019-11-22:
Hello,

I am running motioneye on a laptop with 2 graphics cards (intel and nvidia). For as far as i know i uninstalled and disabled the intel one for as far as i can.

Still, i am getting sub-par encoding results on this Quadro M1000M (around 10-15 fps ) when recording (the right fps counter) and there are alot of dips and freezes in the recordings when playing them back.

I tried messing around in the config files of both the camera's and the one of motion to put the ffmpeg codec to 'mp4:h264_nvenc' but this does nothing, there is still 0% usage on the nvidia GPU.

Am i doing something wrong?

Here is a screenshot of the usage (in glances):
[image]

jasaw, 2019-11-24:
Find out whether your ffmpeg supports h264_nvenc. Run something like ffmpeg -codecs | grep h264_nvenc.

Maxloader2000, 2019-11-25:
I think my ffmpeg does, I am getting this back:
```
ffmpeg version 3.4.6-0ubuntu0.18.04.1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.3.0-16ubuntu3)
  configuration: --prefix=/usr --extra-version=0ubuntu0.18.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
 DEV.LS h264                 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 (decoders: h264 h264_crystalhd h264_v4l2m2m h264_vdpau h264_cuvid ) (encoders: libx264 libx264rgb h264_nvenc h264_omx h264_v4l2m2m h264_vaapi nvenc nvenc_h264 )
```

jasaw, 2019-11-25:
It's probably a problem with motion software. You'll have to report it to [Motion Project](https://github.com/Motion-Project/motion/issues).

closed 2021-03-05 by starbasessd