motioneye-project/motioneye #1558: Using motioneye as server with more than 30 cameras

CaptainDario, 2019-12-02:
Hello, 
first of all thank you for the unbelievable piece of software!
Currently I am running motioneyeOS on a Raspberry pi 2 as a server with three IP cameras attached to it. This was my setup for testing purposes.
My final setup will have more than 20 (or even 30) cameras which I want to detect motion on. Therefore I will be building a new rig for the server to run on.
My Question now is which part(s) of my server are the most important for motioneye and its motion detection. Is the GPU or CPU more important.

your help would be very much appreciated,
CaptainDario

jasaw, 2019-12-03:
CPU is important as most things are done in the CPU. MJPEG live stream requires JPEG encoding which is done in CPU. Motion detection done in CPU too. Video recording which needs to encode video *may* be offloaded to the GPU but it is not tested, and not officially supported. Maybe someone who has tried video encoding on GPU can chime in?

CaptainDario, 2019-12-06:
what kind of CPU would be necessary to run the motion detection on approximately 20 cameras. Most of them would be around fhd.
Would an ryzen 5 with 12 threads be a feasible choice?

DubhAd, 2019-12-10:
I have a camera that supports both a low resolution and high resolution stream. I'm running the motion detection on the low resolution stream, recording the high resolution stream (and using movie_passthrough). That consumes about 5% to 10% of one core on an i5-8250U.

CaptainDario, 2020-03-18:
Thank you.
I will try to upgrade to something similar.

closed 2020-03-18 by CaptainDario

ministryofsillywalks, 2020-05-10:
> 
> 
> I have a camera that supports both a low resolution and high resolution stream. I'm running the motion detection on the low resolution stream, recording the high resolution stream (and using movie_passthrough). That consumes about 5% to 10% of one core on an i5-8250U.

Sorry for gravedigging but how exactly did you set this up to use the second low res stream for detection and then have record from the high res stream?

zagrim, 2020-05-10:
@ministryofsillywalks check out Motion options `movie_passthrough` and `netcam_highres` https://motion-project.github.io/motion_config.html#movie_passthrough