motioneye-project/motioneye #1559: I need better than 800x480 camera resolution

nigeljohnson73, 2019-12-03:
I'm running 20190911 on a raspberry pi zero with the camera module version 2.1, and I cannot get the camera to stream with a resolution higher than 800x480. Looking through the `dmesg.log` file, when it fails it says it's out of memory.

```
[  918.152915] Out of memory: Kill process 1800 (motion) score 771 or sacrifice child
[  918.152969] Killed process 1800 (motion) total-vm:284464kB, anon-rss:191428kB, file-rss:8kB, shmem-rss:0kB
[  918.281839] oom_reaper: reaped process 1800 (motion), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
```

In fact I was noticing that the board would completely lock up every so often so I bumped the resolution back to 640x480 and it seems to stream a bit better, but I'm guessing the video creation bit is too taxing as it will also lock up about when it should be completing the video, and the videos don't seem to render in browser, and won't play when downloaded either.

So, I'm guessing that the Pi Zero is not suited for anything but streaming really pants resolution video.

My question then, is, as Pi 4 will have about 3 times the CPU so should be less fally overy when doing video completion, but if I get the 1GB version it will still be crap resolution. Would the 4GB version give me some decent 1080p images? It's an expensive option but if it works it saves me about 15 quid on an actual network camera that will solve my issues and it's PI based :)

Any help much appreciated.

jasaw, 2019-12-04:
Minimum motion frames and capture before config options that take up a lot of RAM. Try reducing those and you won't run out of memory.
Pi zero W won't have enough grunt to support high frame rate above 1024x768 resolution

nigeljohnson73, 2019-12-08:
Thanks, I can't get the Zero up high enough, but it is a great fast network cam for monitoring at 1920x1080 and about 20 fps.. it seems to do 30fps but does get a bit jerky occasionally. I've been trying a Pi4 and it's really struggling to capture video once the motion is detected and it's still got loads of CPU and RAM free.. but, that's a topic for another thread :)

closed 2019-12-08 by nigeljohnson73