motioneye-project/motioneye #1561: restart event numbering (%v) at day change

bkffld, 2019-12-08:
Hello,
I'm using MotioneyeOS on PI's quite a while, and I'm extremely satisfied with, thank you for your work!

I'm just looking for a tip on how to restart the event numbering (%v) on midnight. Is there a configuration item I have not found so far or an entry in a config file elsewhere?

Thanks in advance and best Regards,
Bernd

bkffld, 2019-12-15:
A "S85motioneye restart" via cron at midnight will make it, but probably there is an easier and more elegant method?

ccrisan, 2019-12-15:
Just slightly better but still far from elegant: `killall motion` via cron :)

bkffld, 2019-12-16:
Hi crissan,

thank for this hint. I tried it and it's much faster than "S85motioneye restart". 
May be the counter restart at midnight is an option for later versions. As I use a seperate directory for each day (%Y%m%d/%v-%H%M%S) this is a ideal way to get pictures sorted.

Thank you very much for your work on Motioneye, which I greatly appreciate!

Have a nice Christmas season,
Bernd (bkffld)

closed 2021-03-05 by starbasessd