motioneye-project/motioneye #1566: Add functionality to web API

javirs85, 2019-12-13:
I have a running motioneyeos and I talk to the web api to start and pause motion detection and to trigger capture from a series of sensors. This works like a charm.

Now I need to add an script in the motioneyeos that I can execute from a remote location. Let me explain it, I want to:

-  put dual boot in my PI with motion eye os and Kodi.
-  Then I want to use IFTTT in my phone so when I connect to my home wifi I calls the motioneyeos API
- and request the execution of an script that changes the default OS and reboots the machine.

As far as I know everything in the list is possible but the API thing .. is there a way to add functionality to the existing API ? I just need to link my script execution to a webhook. 

I would gladly accept any other suggestion not based on motioneyeos api, but it feels the natural way to go

rakeshdas1, 2020-01-05:
What urls did you use for the API requests to pause motion detection? I'm not sure if the complete API has been implemented as outlined in the API section on the wiki

javirs85, 2020-01-06:
I'm not stopping the detection. Right now I have it on and simulate detection via API. This part of the API does work and, according to the internet, pausing detection (not stopping) seams to be implemented. Will be my next try

closed 2021-03-05 by starbasessd