motioneye-project/motioneye #1567: Unable to access motionEye outside of network

theehmfic, 2019-12-13:
I could really use someones help on this. I have scoured the web with no luck. I am unable to access the streaming feed as well as the main interface when I'm outside of my home network, not hooked to wifi. I went through the port forwarding steps numerous times as well as set up a Dynamic DNS service and still can't get anything to load. When I try it try loading for awhile then eventually times out. I'm very new to Linux and motioneye so I've went through every tutorial and forum I could find during the set up process. I have it working just fine if I'm connected to my home network. Do I have to set up port forwarding for both computers I'm using? I'm trying to do a trial run before buying a bunch of raspberry pi's and setting them up. Right now I have a laptop with Ubuntu and motioneye set up on it and a webcam attached. I have another computer with a VM of Ubuntu running motioneye as well. I have set up port forwarding on the laptop's IP address. As stated everything is working fine if I'm connected to home network, I can access it on my phone and computer, but not outside it. I have motioneye version 0.35.1 and motion version 4.0 installed. Any help would be greatly appreciated.

zagrim, 2019-12-14:
It's good that it's tricky, because if your intention is to open things up to the Internet, it also means opening any security vulnerability in the exposed applications/services to the Internet.

That being said, we need more details to be able to assist.
What you basically need to accomplish is to have a port (or ports, if you need to have access to more than one IP address + port pair in the internal network) in your Internet facing firewall opened and set that up to forward traffic to an internal IP+port. To keep things simple, it'd be best to have just one MotionEye acting as a central server so that you'd only need to set up port forwarding to that (I mean this kind of setup: https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios#multiple-devices-with-a-central-server).

zagrim, 2019-12-14:
Where have you tried to set up the port forwarding? What kind of firewall you have? The reasonable thing to do is to try with something minimal and simple first to make sure you've got the DNS and firewall opening working, and then move on to more complex things like port forwarding. Ideally you'd have some dummy service in the firewall enabled and opened to the Internet, so that you could test connecting to it. When it succeeds, you'd know that at least the DNS thing works and you are able to open a port in the firewall properly.

zagrim, 2019-12-14:
For the sake of security, I'd myself want to have [MotionEye running as a normal user](https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User) and enable HTTPS either by running Apache or [Nginx](https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx) in front of MotionEye before exposing them to the wild. That being said, I have no idea how secure MotionEye or Motion generally are, but it's still a good idea to have some extra layers of security to make it harder to exploit eventually surfacing vulnerabilities or to contain the damage that can be caused.

theehmfic, 2019-12-14:
> It's good that it's tricky, because if your intention is to open things up to the Internet, it also means opening any security vulnerability in the exposed applications/services to the Internet. [...]

zagrim, thanks for the reply. I have been able to access the streams through different pc's and devices(smart phone) when hooked to the home network, but once I unhook my phone from wifi it stops working. I'm not sure about the firewall, I haven't set anything specific myself. I have a Netgear CM1000 modem and a Netgear Orbi router. I just went through and assigned static ip's to the 2 computers I'm trying to use and still no luck. I went into my routers settings and followed the port forwarding steps, and set the port to 8081. The set up I have right now, which isn't the final set up, just trying to figure everything out, I have a laptop with Ubunto 18.04 and motion eye on that. I have a usb camera attached to it. I have a VM setup with Ubunto 18.04. I have tried to set up port forwarding with the laptops ip address. I can access the admin and user side of the laptops motioneye on the vm by using laptops ip:8081/ and ip:8765/ respectively. So I'm getting things to talk to each other, I just can't do it outside the network. I may be going about this all wrong and may just need to get a raspi, but I figured doing it this way would be just the same as a test. I could be completely wrong though. When I try to connect outside the network it just times out and says "site can't be reached ip address took to long to respond" Sorry if I'm all over the place, trying to answer all of your questions. Thanks again for the help.

theehmfic, 2019-12-14:
> It's good that it's tricky, because if your intention is to open things up to the Internet, it also means opening any security vulnerability in the exposed applications/services to the Internet. [...]

Zagrim, Is there a step by step guide for the usage scenario you quoted? That's the setup I'm going to be doing, I'm just trying to get the steps figured out now, once I can get the port forwarding thig figured out.

zagrim, 2019-12-14:
It seems that it is indeed the router you should need to reconfigure, since the modem doesn't seem to have any settings for that according to the manual I found online. In the router's port forwarding settings, you'd need to enter the locally accessible IP address and port ([writing this based on this Netgear support site document](https://kb.netgear.com/31069/How-do-I-configure-port-forwarding-on-my-Orbi-system), which BTW says you need to have static IP for the service that is to be forwarded, so good that you already have that set up). Then it should be possible to access the same port (the document doesn't seem to indicate if there's a possibility to configure different port to be used on the Internet side) using the Internet address assigned to you by your ISP (which you probably have figured out, but which can anyway be found out using something like https://www.whatismyip.com/ . And since I assume that the router also acts as the firewall, it should probably take care of opening up the forwarding port to the outside. But I'm not sure about it.

zagrim, 2019-12-14:
It might be good to tackle that centralised setup in a separate Github issue, although I would assume that there are already questions on that topic with also answers, please check both this project as well as motioneyeos project's Github for those. And anyway, that's something you'd want only after getting at least one box reachable from the Internet.

theehmfic, 2019-12-14:
Zagrim, It seems the router under port forwarding settings only lets me enter internal IP addresses. I went to "whatsmyip" and that address for my public ip is obviously way different than my internal and can't be set in the router. I did just see in that netgear support doc you sent that you have to use a reserved ip address, there is a place to put that in. I had just set the static ip addresses through the operating system. I just tried to enter the reserved IP addresses into the area it has and it says invalid destination IP addresses, even though I'm entering the systems static IP. Isn't the reserved IP the static IP I set for the computer? Also as far as ports i used 8081 for both sides of it

zagrim, 2019-12-14:
It is indeed the internal (LAN) IP address you need to enter in the port forwarding settings, the Internet (WAN/public) address will automatically be the one that's seen from the Internet. Sorry if I wasn't clear or got you confused.
I have no experience of Netgear devices, but it might be that you need to ["reserve" the IP address in the router](https://kb.netgear.com/25722/How-do-I-reserve-an-IP-address-on-my-NETGEAR-router) for it to accept it in port forwarding. From that document it seems like you can leave the OS configured for DHCP and then the router will just assign the same reserved IP for that device every time. So, return the OS network config to DHCP, go to the router settings and reserve the IP address the device got via DHCP (you also need the device MAC (hardware) address). Then go to port forwarding setup, enter that same reserved IP address and the port to be forwarded (e.g. 8081). 8081 should be just as fine for the WAN side, I see no reason why the ISP would block that (unless they block inbound access to all or most of the ports, but let's hope they are not that nasty).

theehmfic, 2019-12-14:
> It is indeed the internal (LAN) IP address you need to enter in the port forwarding settings, the Internet (WAN/public) address will automatically be the one that's seen from the Internet. Sorry if [...]

I will give it a try again, but everytime I tried to enter the ip in the reserved field is says it a "invalid destination ip" but I'm entering the static ip of the computer. I will try to change the os ip back to dynamic then try to do the reserve up. I'm not sure why it would say invalid destination. I'll give it a go when I get back in front of the computer and let you know. This is very frustrating and shouldn't be this difficult. It's probably some little feature on the router messing everything up

theehmfic, 2019-12-14:
Zagrim,
 it's not letting me input a reserved ip, for some reason. This has become really frustrating. Everything I have seen for other routers port forwarding is easy, not this one. I went out and bought a raspi4 and I'm going to install motioneyeOS on it and see if I have better luck using that. I did see one setting on the router that would allow me to set a static ip for the internet IP address. I'm assuming this means for the public IP? Is this something I need to do? Another thing I just thought about. I have this modem and router I've been talking about, but there is also a modem/router provided from my isp that is set up on it's own network. I had to leave it on for the comcast security system. Could this somehow be affecting the port forwarding, like a double NAT situation? They are their own networks and I have the two modems hooked up with a cable splitter. Maybe this is the issue?

theehmfic, 2019-12-14:
I looked at the logs from my router, [LAN access from remote] from 31.45.242.224:8902 to 192.168.1.xxx:8081, Saturday, December 14, 2019 06:19:31   There is a bunch of them, every time I've tried to remote in and I have yet to be successful. I don't get it. I'm about to give up on this. I tried unhooking the other modem/router that was in the house just to eliminate a variable and no dice

theehmfic, 2019-12-15:
> It is indeed the internal (LAN) IP address you need to enter in the port forwarding settings, the Internet (WAN/public) address will automatically be the one that's seen from the Internet. Sorry if [...]

Zagrim,
So I had a epiphany,  I have been entering the ip address of the laptop I have with motioneye on and a usb camera, and not the computer that's acting as my server that I have motioneye on. Could this be the issue? I would think I can port forward any computer, especially since I can see it while in network but I cant outside. I'm so confused. Should I just start from scratch, use the raspi4 with a camera, and try to port forward just that, make it super simple until I get it figured out?

CRImier, 2019-12-15:
the "wrong IP" thing is probably not the problem, but simplifying the setup should definitely help.

theehmfic, 2019-12-15:
> the "wrong IP" thing is probably not the problem, but simplifying the setup should definitely help.

I agree. I'm going to start fresh, use just a raspi4 first and try to get port forward working on that and then circle back. I'm assuming doing port forwarding on a single raspi4 is possible?

CRImier, 2019-12-15:
port forwarding is something you do on a router => it should be possible to do PF *for* a single Pi4, yes.

theehmfic, 2019-12-15:
> port forwarding is something you do on a router => it should be possible to do PF _for_ a single Pi4, yes.

I  know its performed on the router, I guess I was meaning are you able to access it from outside network through port forwarding,  but it makes sense it can. I was just thinking it had to be accessed through a different pc the way I was planning. I'll circle back around in the morning after I try to get just one Pi4 linked up with port forwarding

zagrim, 2019-12-15:
> it's not letting me input a reserved ip, for some reason.

@theehmfic are you saying that it still didn't accept the IP for port forwarding server IP? Have you succeeded ever so far to put anything there? It didn't accept the static IP set in the OS, and it didn't accept the IP reserved in the router LAN setup, either? If that is the case, you probably should try to get either more model specific manuals or some help from Netgear support or their community on how to set those things up. Myself I don't know are all the Orbi routers the same or do they come with different firmwares and different ways to set things up.

Cleaning up the setup would probably make it easier for you to concentrate on the essentials, but as long as there's no success at all with connecting from the outside, getting to know how the router really wants port forwarding to be set up is the main goal.

zagrim, 2019-12-15:
> I looked at the logs from my router, [LAN access from remote] from 31.45.242.224:8902 to 192.168.1.xxx:8081, Saturday, December 14, 2019 06:19:31 There is a bunch of them, every time I've tried to remote in and I have yet to be successful.

Is the the latter (internal) IP in the logs in those cases is the one you had entered in router settings? That would suggest that the router had accepted it anyway, but from what you've written I've come to understand that none of the IPs you've tried to enter were accepted. I'm baffled.

theehmfic, 2019-12-15:
> Is the the latter (internal) IP in the logs in those cases is the one you had entered in router settings? That would suggest that the router had accepted it anyway, but from what you've written I've [...]

Yeah that's what I don't get. The second ip is the static one I created in the os and shows when I scan it, and it shows in the port forwarding section of the router to select but the port forwarding just isn't working. I'm baffled too. I think I'm gonna try one more thing and then if it doesn't work tear it all down and make it super simple with a pi4 and try to just port forward that and see what happens. I looked into Netgear support and it says you only have free support for 90 days after purchase, it's been longer than that. Or I can pay $69 for a year of support or $49 for one time. What a joke after I spent $400 bucks for a modem and wifi mesh system.

zagrim, 2019-12-15:
@theehmfic As I still don't know the details of your specific router, I just did some more searching and came to this site (annoyingly "for dummies", but the most detailed one I've seen so far on the topic) that states (emphasis is mine):

> 4.Check the External IP addresses field in the port forwarding rule of Netgear Orbi RBR50 Router
Check if Netgear Orbi RBR50 Router has a field called “External IP address” in the port forwarding section. **Make sure you Enter "0.0.0.0" in the External IP address fields.**
External IP addresses are not entered unless you are restricting access from specific WAN addresses. You also would not enter LAN IPs or your own public IP.

Would that help?

theehmfic, 2019-12-15:
> @theehmfic As I still don't know the details of your specific router, I just did some more searching and came to [this site](https://openmyip.com/Netgear-Orbi-RBR50-router-setup) (annoyingly "for [...]

I checked and there's no section for external ip on the port forwarding page. I have a Orbi RBR40 if that helps. I think it's pretty similar to the RBR50 it's just the newer model. I may look at the firewalls on the linux laptop, maybe that's blocking it? Even though that page said it was for dummies, it had some confusing wording. Just to recap my config, I have motioneye set up on a vm with no camera attached. I have motioneye installed along side ubunto 18.04 with a webcam. I have a static ip set for the laptop with webcam. I have entered the static ip  of the laptop into the port forwarding rules using port 8081 both internal and external on the router. I can access that just fine in my network but not outside. At this point I'm going to try just setting up a Pi4 and port forward that and see what jhappens

zagrim, 2019-12-16:
> I checked and there's no section for external ip on the port forwarding page. I have a Orbi RBR40 if that helps.

Ok, pages describing port forwarding with that specific model indeed don't mention external IP.

> I may look at the firewalls on the linux laptop, maybe that's blocking it?

I would expect not only the destination address but also the source address of incoming request to be rewritten by port forwarding, thus your laptop would only see packets as if they were coming from your router, and never with the any external address. So there shouldn't be any change from the perspective of the laptop networking stack whether the request originated locally or from the Internet. But then again, having the laptop firewall blocking things would perfectly explain the end result, so please go ahead and check that.

Regarding your setup, the only piece I see as potentially problematic is the VM, as it usually requires some extra configuration to get things inside a VM accessible from outside of the VM.

theehmfic, 2019-12-16:
> Ok, pages describing port forwarding with that specific model indeed don't mention external IP.
>
>
> I would expect not only the destination address but also the source address of incoming request to [...]

Ok, so I simplified everything, I set up a PI4, configured static ip for it, set the port forwarding for it on the router and no luck. I set up a Dynamic DNS with the Pi4 IP address, no luck, I reserved the Pi4 static IP address in the router no luck. I tried to set it as default dmz server, no luck. If I have UPnP enabled or disabled still no luck, I tried one of those port checkers and it shows my public IP and says port 8081 is open, so that would mean it ready for business right? If so why can't I tap into it when I'm off the local net. I'm so seriously beyond frustrated at this point. I feel like I must be doing something wrong, but all the tutorials and everything I've read show it as easy as putting th info into the port forwarding section of the router and boom Bob's your uncle, but for me no dice. The only thing I can think, but with my limited knowledge is there is something wrong with the router, but I highly doubt it. I've read about being in a double NAT situation, but that is usually if you have two router in place, correct? I don't have that so I don't know what to do now besides taking a bat to all of it :)

theehmfic, 2019-12-16:
> Ok, pages describing port forwarding with that specific model indeed don't mention external IP.
>
>
> I would expect not only the destination address but also the source address of incoming request to [...]

Maybe I'm just doing something wrong, something simple I'm overlooking.  What are the steps you use to port forward? What port to do you have designated on motioneyeos, 8081? Do you use a DNS and what ip address do you put in there? The ip of the machine your trying to port/access, correct? What ip and port do you put in the browser on outside network? The machine your trying to access and port correct? Or the DNS address? So frustrating!

zagrim, 2019-12-16:
Uh, that's too bad :(
What I would do in that kind of situation is to take Wireshark and start looking if something comes in and if so does the destination service try to respond. But the use of a packet sniffer is already quite hard to try to explain here (not to mention I think it requires a suitable wifi adapter that supports monitor mode).

Myself I don't have (yet) tried to enable access from outside, and since I have a OpenVPN server running at home I might actually not even need to do that. But that's not relevant.

As for the port forwarding, it shouldn't matter what port is forwarded, whatever service has bound to that port should work just the same. And it shouldn't matter if the internal and external ports are the same or differ. The only things that should matter is that the router knows the internal IP and port it should use to forward all traffic that arrives to a port on the external interface. And of course one must use the Internet (WAN) address of the router when connecting, the private address blocks (10.x.x.x, 172.16.x.x - 172.31.x.x, 192.168.x.x) cannot be used on the Internet. But again it doesn't matter if you enter a DNS name or IP address in the browser, the browser will always communicate using the IP address.

theehmfic, 2019-12-16:
> Uh, that's too bad :(
> What I would do in that kind of situation is to take Wireshark and start looking if something comes in and if so does the destination service try to respond. But the use of a packet sniffer is already quite hard to try to explain here (not to mention I think it requires a suitable wifi adapter that supports monitor mode).
> 
> Myself I don't have (yet) tried to enable access from outside, and since I have a OpenVPN server running at home I might actually not even need to do that. But that's not relevant.
> 
> As for the port forwarding, it shouldn't matter what port is forwarded, whatever service has bound to that port should work just the same (if you want the stream be accessible, forward that port, and if you want the MotionEye web UI be accessible, forward that port). And it shouldn't matter if the internal and external ports are the same or differ.
> The only things that should matter is that the router knows the internal IP and port it should use to forward all traffic that arrives to a port on the external interface. And of course one must use the Internet (WAN) address of the router when connecting, the private address blocks (10.x.x.x, 172.16.x.x - 172.31.x.x, 192.168.x.x) cannot be used on the Internet. But again it doesn't matter if you enter a DNS name or IP address in the browser, the browser will always communicate using the IP address.

So when you say use the internet WAN address, what do you mean. When I try to access the port from outside I've been putting 192.168.x.xxx into the browser, this is the address I entered into the port forwarding section of the router. Is that not correct?  For the DNS I have been using No-IP and at first it has the public WAN address the one starting with 72.xxx.xxx.xx but I've been putting the LAN ip 192.168.X.XXX into it, because I thought that was correct. Maybe I've had the wrong I{ addresses the whole time. Ok so what address do I put into the dns field when I set that up? What address do I put into the browser outside the network? Sorry for all the questions, I just want to get this figured out, I've spent way to much time on something this simple.

theehmfic, 2019-12-16:
> Uh, that's too bad :(
> What I would do in that kind of situation is to take Wireshark and start looking if something comes in and if so does the destination service try to respond. But the use of a [...]

I finally got it figured out!  I think there was two issues. I have a WDCloud on my network and it's operating on port 80 when UPnP is enabled, I disabled UPnP. The second, when I was entering the Dynamic DNS address I was entering it, address.mynetgear.com instead of http://address.mynetgear.com:80 and 8081 respectively. I assumed since i had the port forwarding with those ports designated i didn't realize i had to put them in the address as well. For some reason i assumed the DNS picked it up through the port forwarding on router, not sure why I thought that. The only examples I've seen in tutorials is to just put the address in the browser and never mentioned adding ports like you do when your doing it internally.  I found a obscure 1 paragraph how-to on no-IP website. So it's all good now. My only question now is can change the port that's designated in the http section on motioneyeos designating port 80? I'd like to still have my cloud backups for my files for home network. Thanks for all the help Zagrim.

zagrim, 2019-12-17:
> What address do I put into the browser outside the network?

I don't know if you figured this out already, but in the browser you put the IP address your ISP has assigned to you (to the router, more specifically), which is the one shown by e.g. https://www.whatismyip.com/ , and the port should be whatever you've put in External Port in the port forwarding. 

As for the rest of your questions, I don't really understand. My MotionEye UI is found from port 8765 which I think is the default, and the streaming ports are 80xx. If MotionEyeOS binds itself to port 80, I don't know about that since I've never used it, only ME as a standalone application. And I don't know what WDCloud is or how it works.
But anyway, if you want to port forward something to port 80 in the Raspi4 with still allowing something else to use port 80 (on some other device within the LAN), you just enter the IP of the raspi4 as the internal IP, port 80 as the internal port, and something else than 80 (say, 8080) to the external port in port forwarding settings. Then you should be able to reach raspi4:80 from youpublicinternetaddress:8080.

So, to yet recapitulate: 
The internal IP and internal port in port configuration should match your "target" inside your wifi. 
The external port in port forwarding configuration should be whatever you want to use when accessing the "target" from outside (i.e. the Internet).
When accessing forwarded service from the Internet, you should use the routers WAN address (not any of the internal addresses) and the external entered in the port forwarding configuration.

And with browsers the port 80 is the default (since that is the default port for HTTP protocol), for accessing any other port it needs to explicitly be speficied in the URL.

theehmfic, 2019-12-17:
> > What address do I put into the browser outside the network?
> 
> I don't know if you figured this out already, but in the browser you put the IP address your ISP has assigned to you (to your router, more specifically), which is the one shown by e.g. https://www.whatismyip.com/ (when opened from within your wifi, I think you said it was 73.x.x.x), and the port should be whatever you've put in External Port in the port forwarding.
> 
> As for the rest of your questions, I don't really understand. My MotionEye UI is found from port 8765 which I think is the default, and the streaming ports are 80xx. If MotionEyeOS binds itself to port 80, I don't know about that since I've never used it, only ME as a standalone application. And I don't know what WDCloud is or how it works.
> But anyway, if you want to port forward something to port 80 in the Raspi4 with still allowing something else to use external port 80 (on some other device within the LAN), you just enter the IP of the raspi4 as the internal IP, port 80 as the internal port, and something else than 80 (say, 8080) to the external port in port forwarding settings. Then you should be able to reach raspi4:80 from youpublicinternetaddress:8080. Edit: but is your WDCloud thingie even accessible from the Internet? If not, then it shouldn't matter at all here. You can have many services listening on port 80 inside your network just as long as they have separate IP addresses.
> 
> So, to yet recapitulate:
> 
> * The internal IP and internal port in port configuration should match your "target" inside your wifi (e.g. your raspi4 IP and port 8081 for camera video stream from MotionEye).
> * The external port in port forwarding configuration should be whatever you want to use when accessing the "target" from outside (i.e. the Internet). It can be the same as the internal port, or not, that's up to you to decide, but you need still to have separate external ports for separate internal forwarding targets.
> * When accessing the forwarded service from the Internet, you should use the routers WAN address (not any of the internal addresses) and the external port entered in the port forwarding configuration.
> 
> And with browsers the port 80 is the default (since that is the default port for HTTP protocol), for accessing any other port it needs to explicitly be specified in the URL.

Ok that makes sense. Now that I have the port forwarding figured out I'm going to start my server up that runs Unbuntu and put motion on it and get the 2 Pi4's talking to it, then set that server up to be accessible from the outside. I think what was hanging me up was having the static IP, Dynamic DNS and port forwarding not on the same page. I didn't really realize till you just stated that 8081 or any 8xxx port is just an extension of port 80, but that makes sense now. The WDCloud is just a Western Digital hard drive that sits on your desk and is accessible from outside  like on your phone through a app. You can interface to it through your pc by either pulling it up through ip in browser or using a pc program that comes with it.  Basically a NAS for dummies, so it's perfect for me! Thanks again for the help. I'll circle back if I get tripped up anymore.

closed 2019-12-18 by theehmfic

zagrim, 2019-12-18:
Great that you got it working! (and thanks for closing the ticket)

jakepals, 2020-02-16:
I use services like ZeroTier and remote.it so that I don't need to open up firewall ports. See if this helps.

You can refer to documentation on these sites:
https://discourse.pi-hole.net/t/how-to-easily-use-your-pi-hole-outside-of-your-personal-network/18878
https://www.raspberrypi.org/documentation/remote-access/access-over-Internet/README.md