motioneye-project/motioneye #1576: Unable to configure Hiseeu IP camera

Tokn59, 2019-12-17:
Hello, Many thanks for making this great software available to the community !!!

I am trying to add an IP camera (latest build of MEOS). Details of the camera : 

- Hiseeu HSY-HB612-P
- Works perfectly in Blue Iris, Shinobi via ONVIF protocol
- Details CODEC see attached image

I have tried many settings in ME but no luck so far. Any suggestions ?

Many thanks !!!

[image]

Tokn59, 2019-12-21:
Additional comment : on the website of the cameav make it is recommended to use this information : 
 
rtsp://admin:password@192.168.1.206//user=admin_password=password_channel=1_stream=0.sdp

Cannot get this to work ...

HJSK, 2020-04-01:
Hi Tokn, 
Try this one : rtsp://192.168.1.141:554/user=admin_password=tlJwpbo6_channel=1_stream=0.sdp?real_stream

Let me please know it worked
succes!

Swekkerooni, 2020-12-13:
> Hi Tokn,
> Try this one : rtsp://192.168.1.141:554/user=admin_password=tlJwpbo6_channel=1_stream=0.sdp?real_stream
> 
> Let me please know it worked
> succes!

I'm just stumbling in from Google, but this worked for me! Thank you, HJSK!

closed 2021-03-05 by starbasessd