motioneye-project/motioneye #1577: Unable to set the amount of pics sent to cloud...

njordan77, 2019-12-17:
Is there any settings that allows to select the amount of pics to be sent to cloud (dropbox).

I've tried a lot but whatever is change in motion detection does not change anything...right now i receive one pic per second and it stars arround 10secs before the motion starts...so its already 10pics with no valuable information,.....huge overhead.

I tried these, especially the Before and After frames...but not sure if a setting of 1 frame means one picutre......

overall i have camera set to 15fps
Thanks a lot Norbert

Motion Gap | seconds |  
-- | -- | --
Captured Before | frames |  
Captured After | frames |  
Minimum Motion Frames | frames | ?

zagrim, 2019-12-18:
Are you asking for a feature just to fix bad configuration?

Yes, frames equal to pictures, either as still images or within a video. To get the number of frames that matches some value in seconds, you just need to multiply the number of secods by the frames per second (fps). 15 fps means 15 frames in a second, and 10 frames in two thirds of a second.

If you have no wish to have any pics before the detected motion starts, just set "Capture Before" to zero.

zagrim, 2019-12-18:
Just as a note: depending on the environment the camera is pointing at and the events that occur there, it might be hard to get the motion detection "just right". I would be extremely happy if I had just 10 frames of non-movement per captured video from my cams outside of the house... :)

Please see [Motion configuration docs](https://motion-project.github.io/motion_config.html#OptTopic_Motion_Detection) for some more detailed help on the options that MotionEye sets and how Motion uses them. Most of the detection options in ME user interface match 1:1 the corresponding Motion options, some (like threshold and smart mask speed are shown a bit differently by ME).

njordan77, 2019-12-18:
Thanks, regarding "better more than no detection at all" - generally you are right but as dropbox remotely is not too fast i normally have 1920x1024 pics arround 30-40 per trigger, and in the end the information i want to see is maybe 10.

what i do not get is that 15fps would be according to your math 15 pics in a sec....so what does e.g. 10 frames before trigger mean.....10 pics....so this would cover less than a second. Which seems to not be the case as i can see from the pics taken that the time per pic is 1sec....

I will now try to set it all to zero to see if it has any impact. From what i have experimentally figured out is that there is no change if i set the frame before to 30 or to 5. same amount of pics sent to the cloud before the triggering event. Thanks/Cheers Norbert

zagrim, 2019-12-19:
Do you have "Frame Rate" set to 15 in MotionEye under Video Device, or perhaps to 10? The fps setting on the camera doesn't dictate the fps of content passed to motion detection.
What kind of device you are running MotionEye on, by the way?

closed 2021-03-05 by starbasessd