motioneye-project/motioneye #1579: Problem adding IP camera - Noob level

dluca22, 2019-12-21:
Hi folks, i'm here asking for help [(image 1)](https://i.imgur.com/Apmh67m.png)

&nbsp;

I'm having problems adding [this](https://i.imgur.com/fOPcRfP.jpg) specific ip cam bought on [Amazon](https://i.imgur.com/zL6zn2P.png) to MotioneyeOS on Rpi3b+

&nbsp;

I only used MotioneyeOS with wired cameras, both the Rpicam and an USB camera, so my knowledge is very limited.
I read in the amazon specs and in the camera box that this is a P2P IP camera, i don't know if this means something, i tried inserting the IP and get the first image error **port 80: Connection refused** as in [(image 1)](https://i.imgur.com/Apmh67m.png)

&nbsp; 

The camera works fine, it is listed as working with [these programs](https://i.imgur.com/8BD9zSe.png) and works fine in android and [windows](https://i.imgur.com/naJ5K79.png), but i wanted a stand-alone setup i can manage via web gui and could save directly in one spot ("my" setup would be 4 of those cameras)

&nbsp;

I tried tweaking all the things i knew about and got no luck.

Unfortunately my knoledge level is low, so pleas bear with me, i'll insert more pics if needed

&nbsp;

Edit: Just to clarify it's not that I myself want to do this project with this specific camera, but my dad's lately has been targeted by thieves, and he thinks he needs this specific tiny camera to surveille the area without the cameras being seen.

&nbsp;

Unfortunately i was the chosen one to have the task of creating the solution to this setup (because he saw me use it before) but between my job and my daughter i can't seem to find any solution in a short time, and my abilites are really limited, still i'm trying to not disappoint my old man, he really deserves more.

2papercuts, 2019-12-30:
Does the camera have a web page you can access to configure it?

dluca22, 2019-12-30:
no, i can only configure it using the suggested app.
writing its IP address in browser gives me unexisting page

how can i know if it streams RTSP or MJPEG videos, and how can i set its IP in the app? 
i mean, what physiscally do i have to write to test it? i only knew it was something like

`http://192.168.x.xxx `

or something, but i see elsewhere something like

`rtsp://192.168.x.xxx:xxxx/stream`

also, i don't know where to retrieve the right port or how to change it since :80 doesnt work

2papercuts, 2020-01-07:
I only have experience configuring cameras through the provided web pages.
What's the name of the app that you are using to configure it?

closed 2021-03-05 by starbasessd