motioneye-project/motioneye #1580: [Motioneye][ubuntu 19.04] Update not updating [0.40 -> 0.42]

matrix866, 2019-12-21:
Hello,
I was trying to update motioneye from

motionEye Version | 0.40
-- | --
Motion Version | 4.1.1
OS Version | Ubuntu 19.04

To latest 0.42.

I do the
- pip install --upgrade motioneye and it upgraded

 I also did
- systemctl restart motioneye

I also restarted ubuntu but in the web UI I still see old version.

Any hep? Thank you.

mcmidget, 2020-02-20:
```bash
$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux bullseye/sid
Release:        unstable
Codename:       sid
```

```bash
$ pip list | grep motioneye
motioneye                     0.41
``` 
 
```bash
$ pip install motioneye --upgrade
Collecting motioneye
Requirement already satisfied, skipping upgrade: tornado<6,>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye) (5.1.1)
Requirement already satisfied, skipping upgrade: pillow in /usr/local/lib/python2.7/dist-packages (from motioneye) (5.3.0)
Requirement already satisfied, skipping upgrade: jinja2 in /home/johndoe/.local/lib/python2.7/site-packages (from motioneye) (2.8)
Requirement already satisfied, skipping upgrade: pycurl in /usr/local/lib/python2.7/dist-packages (from motioneye) (7.43.0.2)
Requirement already satisfied, skipping upgrade: backports-abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (0.5)
Requirement already satisfied, skipping upgrade: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.4.0.3)
Requirement already satisfied, skipping upgrade: futures in /usr/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.3.0)
Requirement already satisfied, skipping upgrade: MarkupSafe in /home/johndoe/.local/lib/python2.7/site-packages (from jinja2->motioneye) (0.23)
Requirement already satisfied, skipping upgrade: six in /home/johndoe/.local/lib/python2.7/site-packages (from singledispatch->tornado<6,>=3.1->motioneye) (1.5.2)
Installing collected packages: motioneye
  The script meyectl is installed in '/home/johndoe/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.                   
Successfully installed motioneye-0.42
``` 

```bash
pip list | grep motioneye
motioneye                     0.42
```

```bash
$ sudo systemctl restart motioneye
``` 
Login in ME web interface at the configuration section still reads:

	motionEye Version	0.41
	Motion Version	4.2.2
	OS Version	Debian unstable

ccrisan, 2020-02-20:
@mcmidget `$ pip install motioneye --upgrade` indicates that you upgraded motionEye as regular user. However your `systemctl` command most likely starts the system-wide motionEye installation. I'd recommend trying to upgrade with `sudo`.

mcmidget, 2020-02-20:
Oops.. sorry for the noise.
Your tip worked like a charm, thanx!

```bash
$ sudo pip install motioneye --upgrade
[sudo] password for johndoe: 
Collecting motioneye
  Downloading https://files.pythonhosted.org/packages/48/03/fdfeb93cb6e6540c20910d248095603051162ee1a4738e73b7bb43608777/motioneye-0.42.tar.gz (757kB)
    100% |████████████████████████████████| 768kB 713kB/s 
Requirement already satisfied, skipping upgrade: jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye) (2.8)
Requirement already satisfied, skipping upgrade: pillow in /usr/local/lib/python2.7/dist-packages (from motioneye) (5.3.0)
Requirement already satisfied, skipping upgrade: pycurl in /usr/local/lib/python2.7/dist-packages (from motioneye) (7.43.0.2)
Requirement already satisfied, skipping upgrade: tornado<6,>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye) (5.1.1)
Requirement already satisfied, skipping upgrade: MarkupSafe in /usr/local/lib/python2.7/dist-packages (from jinja2->motioneye) (0.23)
Requirement already satisfied, skipping upgrade: backports-abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (0.5)
Requirement already satisfied, skipping upgrade: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.4.0.3)
Requirement already satisfied, skipping upgrade: futures in /usr/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.3.0)
Requirement already satisfied, skipping upgrade: six in /usr/local/lib/python2.7/dist-packages (from singledispatch->tornado<6,>=3.1->motioneye) (1.5.2)
Building wheels for collected packages: motioneye
  Running setup.py bdist_wheel for motioneye ... done
  Stored in directory: /root/.cache/pip/wheels/d1/c4/92/8be8144ebd0e3f3179022bc61b7e5960421c0ba26a69f03ffe
Successfully built motioneye
Installing collected packages: motioneye
  Found existing installation: motioneye 0.41
    Uninstalling motioneye-0.41:
      Successfully uninstalled motioneye-0.41
Successfully installed motioneye-0.42
```

```bash
$ sudo systemctl restart motioneye
```

Now the info at configuration section reads:

    motionEye Version	0.42
    Motion Version	4.2.2
    OS Version	Debian unstable

closed 2021-03-05 by starbasessd