motioneye-project/motioneye #1581: Invalid configuration file line: motion_binary and I cannot access the pi afterwards

javirs85, 2019-12-22:
I'm sure it is the motioneye os including some weird character in my motioneye.conf file. It happens al the time when setting WPASupplicant. 

The problem is that even connecting a HDMI cable and a keyboard to the supposedly headless pi.. I discover I cannot access it. It throws this critical error on boot and stays there. Is not frozen, I can type but I cannot login or make anything.. 

My motioneyeos is crammed with python scripts to read from a PIR sensor and custom scripts to send pushbullet notifications on detection so .. re installing is not my preferred solution .. 

Any clue on how to externally edit (or reset) the motioneyeos.conf from outside (I´m in a W10 machine)

zagrim, 2019-12-23:
Have you looked at MotionEye #1517 or MotionEyeOS #2147?

And for what it's worth, asking in [motioneyeos](https://github.com/ccrisan/motioneyeos) project might be more appropriate.

javirs85, 2019-12-23:
Yes I did and the suggestion there seams to be to edit the motioneye.conf file. My question is how to get to it. I cannot plug the SD in my win10 machine. I cannot SSH the pi, I cannot even plug an HDMI monitor and a USB keyboard.. I have no access to it :(

zagrim, 2019-12-30:
I would expect the only way to resolve this by modifying files on the SD if the Pi won't boot now. Why can't you plug the SD card in another computer?

javirs85, 2019-12-30:
Because I'm on Windows machine :( when I plug the SD I just see one small
partition, I guess boot.

I ended up re installing and doing my scripts again ... This time I made
copies on GitHub for everything

perttiov, 2020-01-03:
Hey I am a beginner.
I noticed there that wpa_supplicant has a different name when editing it.
An alphanumeric number that is not in the original name will be saved.
I got motioneye to work on wifi started working when I removed these additions.
Sorry this translated google translater

closed 2021-03-05 by starbasessd