motioneye-project/motioneye #1583: Still high cpu usage

ycodryn, 2019-12-29:
I keep returning to motioneye to see if high cpu usage was resolved but is not. On my synology docker with 4 cameras my cpu is around 70% and that is only for view. No record, motion or anything enabled. With shinoby, having recording and motion enabled my cpu is around 4-10%. What shinoby has and motioneye not?

jasaw, 2020-01-06:
MotionEye uses motion software to do motion detection. Motion software decodes the camera stream, encodes it to MJPEG (on the CPU), and serves the MJPEG stream. This MJPEG encoding will load up the CPU. What is your CPU load when it is not live streaming? i.e. close your web browser, no MJPEG stream.

ycodryn, 2020-03-18:
Around 70% without live streaming.

ycodryn, 2020-11-06:
8 months passed, remade the docker, same issue. Motioneye using 40-50% of cpu and @800mb RAM and Zoneminder using 20% od cpu @2GB RAM.  Using netcam_decoder h264_qsv does not change usage at all.

closed 2020-11-15 by ycodryn