motioneye-project/motioneye #1584: can't do much in setup_mode

Zixim, 2019-12-29:
this is all i get at http://ip:8080/

[image]

I get the feeling I'm missing something ?

jasaw, 2020-01-06:
try http://ip:8765 ?
Check your `/etc/motioneye/motioneye.conf` file, find which port number has been configured. Default port number is 8765 I think.

Zixim, 2020-01-06:
port 8765 runs fine, but that is beside the point.
I'm trying to get into setup_mode , as per [the wiki](https://motion-project.github.io/motion_config.html#setup_mode).

kleini, 2020-01-06:
@Zixim Are you talking about motioneye or motion? Your link to the wiki points to an option of motion not motioneye. motioneye uses motion underneath but not motions setup_mode.

closed 2020-01-06 by kleini