motioneye-project/motioneye #1586: Motion Eye or Webcam Never Capture Human Movement

ukmercanary, 2019-12-29:
Hello

So I have been using MotionEye with my OLD Foscam and to be honest I am thinking about finding a new solution.  I have spend months trying to get the camera and it's software to track movement.  The camrea is mounted outside my house looking down onto my front door.  However it just records random rubbish. For example it will record when there is a change in light, i.e at sunrise and sunset.  It does not record humans walking up to my front door.  At the moment it just acts as a deterant as knowbody knows it does not work...

Any ideas?  does anyone recommend a camera that they know works? I am almost tempted to get a google Nest cam as my brother has one at they just work.

jasaw, 2020-01-06:
Enable "show debug frames" and it will add "number of changed pixels" on the top right corner of the video. It will also draw a red box around the area with detected motion.
Play around with minimum motion frames and motion threshold. Enable despeckle filter, noise filter. Find out what combination motion detection parameters work for you.

ukmercanary, 2020-01-06:
It is also working alot better now thanks

closed 2021-03-05 by starbasessd