motioneye-project/motioneye #1596: %f in web hook results in empty string

wimdecorte, 2020-01-03:
I have this web hook used when motion is detected:
Using a POST (query)
https://myserver.com/api/webhook/v1/motioned/catch?file=%f&host=%{host}

The web hook api sees the incoming as:

[image]

As per the motion documentation (https://motion-project.github.io/motion_config.html#conversion_specifiers) I would expect %f to give the full file name and path.
If I change "f" to "F" I get the part of the file name without path or extension.

wimdecorte, 2020-01-03:
This is with:
motionEye Version	0.42
Motion Version	4.2.2
OS Version	Raspbian 10

ccrisan, 2020-01-25:
That's using `on_event_start` behind the scenes which doesn't accept `%f`. At that time, the file might not be created at all, let alone being fully written. Also, it might be a movie, a jpeg, both, or none.

See if *File Storage > Call A Web Hook* works for you. Worst case scenario, use *File Storage > Run A Command* and provide your own shell script.

closed 2020-01-25 by ccrisan